ARMmbed/mbed-os #15186: shared_highprio_event_queue

tiger902, 2021-12-07:
i run mbedos 6.15 on nucleo_l4p5zg, and  occurs the following errors. how can i avoid it, thks.

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: 00AA8CD6
R   1: 2000C2EC
R   2: 20002BB8
R   3: 20002BA0
R   4: 00000000
R   5: 00000000
R   6: 00000000
R   7: 00000000
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 00000000
SP   : 20002C08
LR   : 08033D71
PC   : 0802D874
xPSR : 01000000
PSP  : 20002BE8
MSP  : 200177D8
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
BFAR : 00AA8CEA
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x802D874
Error Value: 0x2000DE3C
Current Thread: shared_highprio_event_queue Id: 0x20002DC0 Entry: 0x8015DC9 StackSize: 0x400 StackMem: 0x20002948 SP: 0x20002C08
Next:
shared_highprio_event_queue  State: 0x2 Entry: 0x08015DC9 Stack Size: 0x00000400 Mem: 0x20002948 SP: 0x20002B00
Ready:
main  State: 0x1 Entry: 0x0805274D Stack Size: 0x00001000 Mem: 0x20002E48 SP: 0x20003CE0
nanostack_event_thread  State: 0x1 Entry: 0x08030271 Stack Size: 0x00001800 Mem: 0x2000C358 SP: 0x2000DAD0
rtx_idle  State: 0x1 Entry: 0x08077FED Stack Size: 0x00000380 Mem: 0x200162D8 SP: 0x20016618
Wait:
rtx_timer  State: 0x83 Entry: 0x08079219 Stack Size: 0x00000300 Mem: 0x20016658 SP: 0x200168C0
Delay:
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_L4P5ZG
-- MbedOS Error Info --

my mbed_config.h as follow:

/*
 * mbed SDK
 * Copyright (c) 2017 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
[... 395 lines cut ...]
#define MBED_LFS2_CACHE_SIZE                                                  64                                                                                               // set by library:littlefs2
#define MBED_LFS2_ENABLE_INFO                                                 0                                                                                                // set by library:littlefs2
#define MBED_LFS2_INTRINSICS                                                  1                                                                                                // set by library:littlefs2
#define MBED_LFS2_LOOKAHEAD_SIZE                                              64                                                                                               // set by library:littlefs2
#define MBED_LFS_BLOCK_SIZE                                                   512                                                                                              // set by library:littlefs
#define MBED_LFS_ENABLE_INFO                                                  0                                                                                                // set by library:littlefs
#define MBED_LFS_INTRINSICS                                                   1                                                                                                // set by library:littlefs
#define MBED_LFS_LOOKAHEAD                                                    512                                                                                              // set by library:littlefs
#define MBED_LFS_PROG_SIZE                                                    64                                                                                               // set by library:littlefs
#define MBED_LFS_READ_SIZE                                                    64                                                                                               // set by library:littlefs
#define MBED_STACK_DUMP_ENABLED                                               0                                                                                                // set by library:platform
#define MBED_TRACE_COLOR_THEME                                                0                                                                                                // set by library:mbed-trace
#define MEM_ALLOC                                                             malloc                                                                                           // set by library:mbed-trace
#define MEM_FREE                                                              free                                                                                             // set by library:mbed-trace
#define PPP_DEBUG                                                             0                                                                                                // set by library:ppp
// Macros
#define MBEDTLS_CIPHER_MODE_CTR                                                                                                                                                // defined by library:SecureStore
#define MBEDTLS_USER_CONFIG_FILE                                              "mbedtls_wisun_config.h"                                                                         // defined by application
#define NSAPI_PPP_AVAILABLE                                                   (MBED_CONF_PPP_ENABLED || MBED_CONF_LWIP_PPP_ENABLED)                                            // defined by library:ppp
#define NSDYNMEM_TRACKER_ENABLED                                              MBED_CONF_NANOSTACK_LIBSERVICE_NSDYNMEM_TRACKER_ENABLED                                          // defined by library:nanostack-libservice
#define NS_USE_EXTERNAL_MBED_TLS                                                                                                                                               // defined by library:nanostack
#define UNITY_INCLUDE_CONFIG_H                                                                                                                                                 // defined by library:utest
#define _RTE_                                                                                                                                                                  // defined by library:rtos

#endif

mbedmain, 2021-12-07:
@tiger902 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

tiger902, 2021-12-07:
Target(s) affected by this defect ?
NUCLEO_L4P5ZG

Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

What target(s) are you using?
NUCLEO_L4P5ZG
What toolchain(s) are you using?
GCC_ARM
What Mbed OS version are you using?
mbed os 6.15
It would help if you could also specify the versions of any tools you are using?
keil 5.35.00
How can we reproduce your issue?
I am unable to resolve this hardware error,

tiger902, 2021-12-07:
What target(s) are you using?
NUCLEO_L4P5ZG
What toolchain(s) are you using?
TOOLCHAIN_ARM
What Mbed OS version are you using?
mbed os 6.15
It would help if you could also specify the versions of any tools you are using?
keil 5.35.00
How can we reproduce your issue?
I am unable to resolve this hardware error,

jeromecoutant, 2021-12-07:
Hi @tiger902 

> Target(s) affected by this defect ? NUCLEO_L4P5ZG

This target is not yet supported in mbed-os :-)
So I suggest you to push a pull request with this support, we will review it and maybe it will solve your issue !
Thx

0xc0170, 2021-12-07:
I'll close this issue as it's not yet supported, the discussion can continue.

closed 2021-12-07 by 0xc0170