ARMmbed/mbed-os #15188: What is the proper way to disable RTOS in mbedOS6.10?

pascalhammer, 2021-12-09:
Hello,

I have seen, that it’s possible to disable RTOS in mbedOS 5.

https://github.com/ARMmbed/mbed-os/issues/7800

Is there a possibility to do this in mbedOS 6.10?

Thanks!

mbedmain, 2021-12-09:
@pascalhammer thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-12-10:
https://os.mbed.com/docs/mbed-os/v6.15/bare-metal/using-the-bare-metal-profile.html - please read the documentation page.

I'll close this as resolved.

closed 2021-12-10 by 0xc0170