ARMmbed/mbed-os #15197: CMake - Custom hardware targets must start with "mbed-" prefix

ladislas, 2021-12-31:
### Description of defect

Custom targets must have an `mbed-` prefix with a name in lower case. We have multiple custom targets for different versions of our hardware.

Using the `mbed-` prefix is strange enough as those targets are not coming to mbed and it is not written in the documentation. It took me quite sometime to understand what was wrong.

In the end we are forced to change the name of our targets from `target_LEKA_V1_2_DEV` to `mbed-leka-v1-2-dev`.

We can live with the `leka-v1-2-dev` but forcing users to format the name of their targets it wrong, the prefix should not be required and if we really can't do without it, documentation should explain the what and why

The process is not as easy 

#### Target(s) affected by this defect ?

n/a

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools 7.49.1

#### How is this defect reproduced ? 

Create a custom target and omit the `mbed-` prefix

0xc0170, 2022-01-10:
> Using the mbed- prefix is strange enough as those targets are not coming to mbed and it is not written in the documentation. It took me quite sometime to understand what was wrong.

Is there an error or warning coming from the tools or our CMake? custom targets should have no requirement like that.

multiplemonomials, 2022-06-10:
I think this requirement is created by this part of the code: https://github.com/ARMmbed/mbed-os/blob/6170f55c645df68bd61d11ff750c3a1605473187/CMakeLists.txt#L189

Mbed wants to derive the CMake target name from the target.json target name, so it converts it to lowercase and prepends mbed- (e.g. NUCLEO_F429ZI -> mbed-nucleo-f429zi).  This means that the CMake target has to start with mbed.