ARMmbed/mbed-os #15198: STM sensor driver LSM6DSLSensor.cpp not compatible with Mbed OS 6

JanneKiiskila, 2022-01-03:
```
Compile [ 99.7%]: LSM6DSLSensor.cpp
[Warning] LSM6DSLSensor.cpp@1108,13: 'void wait_ms(int)' is deprecated: 'wait_ms' is deprecated in favor of explicit sleep functions. To sleep, 'wait_ms' should be replaced by 'ThisThread::sleep_for' (C++) or 'thread_sleep_for' (C). If you wish to wait (without sleeping), call 'wait_us'. 'wait_us' is safe to call from ISR context. [since mbed-os-5.14] [-Wdeprecated-declarations]
[Warning] LSM6DSLSensor.cpp@1108,13: 'void wait_ms(int)' is deprecated: 'wait_ms' is deprecated in favor of explicit sleep functions. To sleep, 'wait_ms' should be replaced by 'ThisThread::sleep_for' (C++) or 'thread_sleep_for' (C). If you wish to wait (without sleeping), call 'wait_us'. 'wait_us' is safe to call from ISR context. [since mbed-os-5.14] [-Wdeprecated-declarations]
```

The driver is not compatible with Mbed OS 6.0. 

Should be trivial to fix, though I'd personally would want to see the driver in GitHub rather than in https://os.mbed.com/teams/ST/code/X_NUCLEO_IKS01A2//file/138a7a28bd21/Components/LSM6DSL.lib/.

(As you can't raise issues there etc.).

Doing a simple ifdef should fix this issue.

```
#IF MBED_OS_VERSION_MAJOR == 5
   wait_ms(10);
#ELSE
   ThisThread::wait(10ms);
#ENDIF
```

Should do the trick. @nikapov-ST

#### Target(s) affected by this defect ?

Any using the sensor LSM6DLSensor.cpp

#### Toolchain(s) (name and version) displaying this defect ?

GCC ARM 
- not compiler specific, any compiler.

#### What version of Mbed-os are you using (tag or sha) ?

b005bf21337e6ffe5393671b92066c3b09c5204b

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

1.10.5

#### How is this defect reproduced ? 

Try compiling https://github.com/AVSystem/Anjay-mbedos-client

mbedmain, 2022-01-03:
@JanneKiiskila thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-01-11:
The driver was updated as I can see. But it might have broken 5.15.x apps (updated to use ThisThread only, I don't think it was propagated  to 5.15 releases).

jeromecoutant, 2022-01-11:
First update done in https://os.mbed.com/teams/ST/code/LSM6DSL/ 
by @cparata

For Github question, thx @avilei

JanneKiiskila, 2022-01-11:
I think it would be easiest for all - if the ifdef (nasty) solution would be used, since that's the ONLY thing that prevents both Mbed OS 5 and 6 compatibility. Now it only works with Mbed OS 6 @nikapov-ST  @cparata.

JanneKiiskila, 2022-01-11:
Plus one additional request - could the main library be updated, too?

https://developer.mbed.org/teams/ST/code/X_NUCLEO_IKS01A2/#138a7a28

To bring in the same change over there as well.

0xc0170, 2022-01-26:
Was everything updated? Can this be closed or what is missing otherwise?

JanneKiiskila, 2022-01-26:
https://os.mbed.com/teams/ST/code/X_NUCLEO_IKS01A2//file/138a7a28bd21/Components/ would need updating, too (just to get them to pull the latest ones).

JanneKiiskila, 2022-01-26:
Excellent, thank you for your support!

JanneKiiskila, 2022-02-07:
Seems I got the version information wrong (the define).

```
#if MBED_MAJOR_VERSION == 5
  wait_ms(10);
#else
  ThisThread::sleep_for(10ms);
#endif
```

It's actually that way around. I swear I could have sworn I tested it, but clearly not with the example code I had. Apologies for that @cparata .

JanneKiiskila, 2022-02-07:
Is there any paln to get Mbed OS drivers for IKS02A1 and IKS01A3?

cparata, 2022-02-08:
Hello,
I fixed the issue on MBED_MAJOR_VERSION for LSM6DSL. IKS01A3 should already be available in mbed OS 6. IKS01A3 mounts newer sensors than the IKS01A2, so I suggest to use IKS01A3 instead of IKS01A2.
Best Regards,
Carlo

JanneKiiskila, 2022-02-08:
Thank you very much @cparata .

closed 2022-02-08 by JanneKiiskila