ARMmbed/mbed-os #15201: Dispatching to a shared event queue from the main thread causes a hard fault

Arachnid, 2022-01-10:
The sample "Shared event example: running the shared queue from main" does not work (at least on a Nucleo-STM32L433; I don't have other hardware to test), and results in a fault.

# Steps to reproduce
 1. Create a project with the code from [this sample](https://github.com/ARMmbed/mbed-os-snippet-Shared_Events_2/blob/v6.15/main.cpp) and `platform.callback-nontrivial` and `events.shared-dispatch-from-application` set to true in `mbed_app.json`.
 2. Compile and run the example

# Expected Result
Output as described in the sample

# Actual result
A hard fault:
```
Starting in context 0x200020B8

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R   0: 00000000
R   1: 20001A10
R   2: 0007D000
R   3: 0000028E
R   4: 20001E78
R   5: 00000000
R   6: 20001EB0
R   7: 00000000
R   8: 00000000
R   9: 00000000
R  10: 00000000
R  11: 00000000
R  12: 20000500
SP   : 2000FF88
LR   : 0801090F
PC   : 20001A10
xPSR : 60000051
PSP  : 20001CF0
MSP  : 2000FF68
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000001
BFSR : 00000000
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
Mode : Handler
Priv : Privileged
Stack: MSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x20001A10
Error Value: 0x20001D14
Current Thread: <unnamed> <handler> Id: 0x20002030 Entry: 0x3E8 StackSize: 0x0 StackMem: 0x1E SP: 0x2000FF88 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_L433RC_P
-- MbedOS Error Info --
```

mbedmain, 2022-01-10:
@Arachnid thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Arachnid, 2022-01-10:
# What target(s) are you using?
NUCLEO-L433RC-P

# What toolchain(s) are you using?
The default arm compiler.

# What Mbed OS version are you using?
6.13.0

# It would help if you could also specify the versions of any tools you are using?
No external tools.

# How can we reproduce your issue?

Please see above.

0xc0170, 2022-01-11:
@ARMmbed/team-st-mcd please review

jeromecoutant, 2022-01-11:
> @ARMmbed/team-st-mcd please review

Hi @0xc0170 
To be honest, I don't think that updating platform.callback-nontrivial and events.shared-dispatch-from-application is linked to ST platforms :-)

I have one comment from
https://github.com/ARMmbed/mbed-os-snippet-Shared_Events_2/blob/v6.15/main.cpp#L9
`SW2` is not a mbed pin name, maybe try `BUTTON1` ?

0xc0170, 2022-01-11:
> To be honest, I don't think that updating platform.callback-nontrivial and events.shared-dispatch-from-application is linked to ST platforms :-)

I was expecting hardware fault rather than a fault in the software. 
@Arachnid would you be able to discover what code is triggering hard fault?

jeromecoutant, 2022-01-28:
@Arachnid 

> I have one comment from https://github.com/ARMmbed/mbed-os-snippet-Shared_Events_2/blob/v6.15/main.cpp#L9 `SW2` is not a mbed pin name, maybe try `BUTTON1` ?

Any update ?
Thx