ARMmbed/mbed-os #15202: Where can I get the spi flash support list of mbed OS?

bigbearishappy, 2022-01-13:
Hello.
I am working with the latest mbed os with my nrf52840 board.
And I want to make my board work with a external spi flash. **I am wondering if there is a support list of spi flash for mbed os?**

What I did:
I have tried the [P25Q16H](https://www.puyasemi.com/uploadfiles/2018/08/20180807152503253.pdf).
It can get the correct vender ID of P25Q16H. But I find something wrong with the mbed os log.
The log shows "Device Does not Have a QE Bit, continue based on Read Inst", but I am sure the P25Q16H does have QE BIT.

mbedmain, 2022-01-13:
@bigbearishappy thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

bigbearishappy, 2022-01-13:
Oh.Sorry.
I will close this issue and create a new one.
Thank you for your reply.

closed 2022-01-13 by bigbearishappy