ARMmbed/mbed-os #15203: Where can I get the spi flash support list of mbed OS?

bigbearishappy, 2022-01-13:
### Description of defect
I am working with the latest [arduino mbed os](https://github.com/arduino/mbed-os) with my arduino nano 33 ble board.
And I want to make my board work with a external spi flash. **I am wondering if there is a support list of spi flash for mbed os?**

#### Target(s) affected by this defect ?
arduino nano 33 ble

#### Toolchain(s) (name and version) displaying this defect ?
I use the arm-none-eabi-gcc in version 10.3.
I download it from [here](https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2)

#### What version of Mbed-os are you using (tag or sha) ?

I use the arduino mbed os with latest commit hash:8ccd98f8d597f315f33fbe943f5a8c5f8d0e28d1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
I use the mbed-cli and it's version is 1.10.5
I install the compile toolchains by following this [page](https://os.mbed.com/docs/mbed-os/v6.15/build-tools/install-and-set-up.html)

#### How is this defect reproduced ? 

I have tried the [P25Q16H](https://www.puyasemi.com/uploadfiles/2018/08/20180807152503253.pdf).
1 make mbed os's qspi can work
2 connec the P25Q16H to the nano 33 ble
3 run the following code to initialize the P25Q16H
`#include "QSPIFBlockDevice.h"
#include "mbed_trace.h"

#define TRACE_GROUP  "main"

using namespace mbed;

QSPIFBlockDevice root(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN, QSPIF_POLARITY_MODE_1, MBED_CONF_QSPIF_QSPI_FREQ);

void setup() {

  Serial.begin(115200);
  while (!Serial);

  mbed_trace_init();

  tr_error("mbed trace test log");
  Serial.println("init");

  root.init();
}

void loop() {

}
`

It can get the correct vender ID of P25Q16H. But I find something wrong with the mbed os log.
The log shows "Device Does not Have a QE Bit, continue based on Read Inst", but I am sure the P25Q16H does have QE BIT.

0xc0170, 2022-01-13:
I dont think there is official list (you can find some libraries on mbed.com portal) but you can find some tested here hal/tests/TESTS/mbed_hal/qspi/flash_configs

bigbearishappy, 2022-01-14:
> I dont think there is official list (you can find some libraries on mbed.com portal) but you can find some tested here hal/tests/TESTS/mbed_hal/qspi/flash_configs

Thank you for you reply.
Now.I am sure that the current mbed os does not support the P25Q16H spi flash.
I am wondering what should I do if I want to make the P25Q16H works fine with mbed os?

0xc0170, 2022-01-14:
You can take an inspiration from the test I shared. And just create your own driver for this flash, as you started. Find out where that warning is coming from and see if you can read/write data.

I haven't used qspi flash for some time so I can't help further here.

bigbearishappy, 2022-01-14:
> You can take an inspiration from the test I shared. And just create your own driver for this flash, as you started. Find out where that warning is coming from and see if you can read/write data.
> 
> I haven't used qspi flash for some time so I can't help further here.

OK. I think I know what should I do now.
Thank you again for your reply~ :-)

closed 2022-01-14 by bigbearishappy