ARMmbed/mbed-os #15208: TDBStore: master record included in RAM table (causing all sorts of problems)

davidsaada, 2022-01-17:
### Description of defect
To start from the end: Problem is that TDBStore holds a pointer to the master record in its RAM table. This can lead to all sorts of issues. Naming a few:
1. Master record key can be caught by iterators (see example code below). This shouldn't happen,as this key is internal.
2. GC process copies all the keys from RAM table to the standby area before areas are flipped. As this includes the master record as a regular one now, storage will gradually get eaten up due to this bug.

Bug lies in [this line](https://github.com/ARMmbed/mbed-os/blob/1443257e40c97141c975f1c5cd7f4cf308a58b5e/storage/kvstore/tdbstore/source/TDBStore.cpp#L948). If we add `_master_record_size`, it will fix the bug (will start iteration on storage from after the master record).

#### Target(s) affected by this defect ?
All targets

#### Toolchain(s) (name and version) displaying this defect ?
All toolchains

#### What version of Mbed-os are you using (tag or sha) ?

Hash is `2a4e48179dd7f67706cbdc7c6bda47d82091c98b`, but this also happens in much earlier versions (5.x).

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Following code, written by @maciejbocianski, iterates over all keys in storage. Iterator catches the master record key as well (which it shouldn't). 
```
kv_iterator_t kvstore_it;
  kv_info_t info;
  int ret = 0;
  size_t kv_store_bytes_used = 0;
  uint16_t key_count = 0;
  char kv_key_out[KV_STORE_MAX_KEY_LENGTH] = {0};
  kv_iterator_open(&kvstore_it, NULL);
  memset(kv_key_out, 0, KV_STORE_MAX_KEY_LENGTH);
  while(kv_iterator_next(kvstore_it, kv_key_out, KV_STORE_MAX_KEY_LENGTH) != MBED_ERROR_ITEM_NOT_FOUND) {
    ret = kv_get_info(kv_key_out, &info);
    if(ret != MBED_SUCCESS) {
      tr_warn("failed to retrieve %s, error: %d", kv_key_out, err_code(ret));
    } else {
      kv_store_bytes_used += info.size;
      tr_debug("key: %s", kv_key_out);
      kv_store_bytes_used += strlen(kv_key_out);
    }
    memset(kv_key_out, 0, KV_STORE_MAX_KEY_LENGTH);
    key_count++;
  }
  kv_iterator_close(kvstore_it);
```
Attempting to run this code on an existing TDBStore will eventuall output the following trace:
```
[WARN][STORAGE]: failed to retrieve TDBS, error: 326
```

mbedmain, 2022-01-17:
@davidsaada thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-01-17:
> Bug lies in this line. If we add _master_record_size, it will fix the bug (will start iteration on storage from after the master record).

@davidsaada Thanks for the report. Will it make sense to create a pull request with the fix?

davidsaada, 2022-01-17:
@0xc0170 Yes, I think so. Didn't check the fix myself though, was just code reading.