ARMmbed/mbed-os #15211: No support for UARTs with shared interrupt on some STM32 targets

amcnicoll, 2022-01-22:
### Description of defect

On some targets (at least STM32L0, STM32G0) with interrupts shared between multiple UART peripherals, it is not possible to use the shared UARTs at the same time. When the second of the shared UARTs calls `serial_irq_set()`, the vector to the first UART's IRQ handler is broken.

I believe this problem was handled for STM32F0 by combining the handler for each peripheral into one (`uart3_8_irq`) and attaching it regardless of whether `serial_irq_set()` is running for UART3 or UART8. Seems like the same thing should have been done for STM32L0 and STM32G0.

If there was a documented user-land workaround for this issue, I wasn't able to find it. Thanks!

#### Target(s) affected by this defect ?

STM32L0, STM32G0

#### Toolchain(s) (name and version) displaying this defect ?

n/a but recent GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

Issue present in `master` as of this writing (1443257e40c97141c975f1c5cd7f4cf308a58b5e)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Discovered while working on a PlatformIO project with the `mbed` framework (Core 5.2.4)

#### How is this defect reproduced ?

Make a project with 2 BufferedSerial objects corresponding to 2 shared UARTs.
For example, this issue was discovered while trying to use USART4 and USART5 on STM32L071.
Suppose you instantiate USART4 first and USART5 second.
The USART5 handler takes over the USART4_5_IRQn.
If data is sent to USART4, the application hangs since the RXNE flag is never cleared by the USART5 handler.

0xc0170, 2022-01-26:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2022-01-28:
Hi

I confirm there is an issue with:
- STM32L0 and USART4_5_IRQHandler
- STM32G0 and USART3_4_LPUART1_IRQHandler/USART3_4_5_6_LPUART1_IRQHandler

Any help is welcomed

amcnicoll, 2022-01-28:
@jeromecoutant, here is how I patched my `serial_device.c` for STM32L0, in case it's helpful until the devs fix this (change extension to *.c)
[serial_device_patch.txt](https://github.com/ARMmbed/mbed-os/files/7960828/serial_device_patch.txt)

jeromecoutant, 2022-01-31:
Thx @amcnicoll 
Do you think you could try to propose a pull request (after pushing your commit in your repo) ?
https://os.mbed.com/docs/mbed-os/v6.15/contributing/index.html

amcnicoll, 2022-02-01:
I've got the commit on a local branch. Am I right in understanding that I must first be granted access to the ARMmbed organization in order to push my branch and create the PR?

0xc0170, 2022-02-01:
> I've got the commit on a local branch. Am I right in understanding that I must first be granted access to the ARMmbed organization in order to push my branch and create the PR?

Create a fork of Mbed Os and create a branch where you can add your commits. You can create a new pull request from your forked repository to this one.

amcnicoll, 2022-02-05:
Thanks, it's up https://github.com/ARMmbed/mbed-os/pull/15221

0xc0170, 2022-02-22:
Fixed via #15221

closed 2022-02-22 by 0xc0170