ARMmbed/mbed-os #15213: LoRa Class C device can't retry-join in OTAA mode

cyliangtw, 2022-01-28:
### Description of defect

Normally, LoRa Class A device could retry join several times in OTAA mode.
As while I set LoRa device as Class C before OTAA handshake `lora.connect()`, if RX1 can't receive 1st join-accept packet, it will stay in the state "`RX2 slot open`" and not proceed any retry-join procedure.

#### Target(s) affected by this defect ?
Not depend on the specific target.
I could reproduce this symptom on both of NUMAKER_IOT_M487 and NUMAKER_IOT_M252.

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

OS v6.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2022-01-28:
@cyliangtw thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Shiny380, 2022-04-27:
@cyliangtw I got around this by joining in Class A mode and then switching to Class C after a successful join. I poked around in the lorawan source for a while to try and fix this but required too many changes in too many places. If I get time soon I might continue with the fix and open a PR but not sure if it's worth it.