ARMmbed/mbed-os #15216: Can't build "Unknow ARM architecture from CRC optimization" Using a Jetson AGX Xavier

MotorCityCobra, 2022-01-30:
### Description of defect
I can't build the library on my Nvidia Jetson AGX Xavier

#### Target(s) affected by this defect ?
```
error: #error "Unknown ARM architecture for CRC optimization"
 #error "Unknown ARM architecture for CRC optimization"
  ^~~~~
storage/kvstore/tdbstore/tests/UNITTESTS/TDBStore/CMakeFiles/tdbstore-unittest.dir/build.make:103: recipe for target 'storage/kvstore/tdbstore/tests/UNITTESTS/TDBStore/CMakeFiles/tdbstore-unittest.dir/__/__/__/source/TDBStore.cpp.o' failed
make[2]: *** [storage/kvstore/tdbstore/tests/UNITTESTS/TDBStore/CMakeFiles/tdbstore-unittest.dir/__/__/__/source/TDBStore.cpp.o] Error 1
CMakeFiles/Makefile2:5584: recipe for target 'storage/kvstore/tdbstore/tests/UNITTESTS/TDBStore/CMakeFiles/tdbstore-unittest.dir/all' failed
make[1]: *** [storage/kvstore/tdbstore/tests/UNITTESTS/TDBStore/CMakeFiles/tdbstore-unittest.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

#### Toolchain(s) (name and version) displaying this defect ?
CMake 3.22.2

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
`cmake .. `runs fine.  
`sudo cmake --build . --config Release`

mbedmain, 2022-01-30:
@MotorCityCobra thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

MotorCityCobra, 2022-01-30:
Sorry for leaving any details out. I updated. I hope that will suffice, please let me know.

0xc0170, 2022-01-31:
If you look at the code closer where it fails:

```
    /* Check to see if we can do assembler optimizations */
#if (defined __GNUC__ || defined __clang__) && \
    (defined __arm__ || defined __ARM_ARCH)
#if (__ARM_ARCH_7M__      == 1U) || \
    (__ARM_ARCH_7EM__     == 1U) || \
    (__ARM_ARCH_8M_MAIN__ == 1U) || \
    (__ARM_ARCH_8_1M_MAIN__ == 1U) || \
    (__ARM_ARCH_7A__      == 1U)
    /* ARM that has Thumb-2 - same unified assembly is good for either ARM or Thumb state (LSRS; IT CS; EORCS reg/imm) */
#define MBED_CRC_ARM_THUMB2     1
#define MBED_CRC_THUMB1         0
#elif (__ARM_ARCH_6M__      == 1U) || \
      (__ARM_ARCH_8M_BASE__ == 1U)
    /* Thumb-1-only ARM-M device - use Thumb-1 compatible assembly with branch (LSRS; BCC; EORS reg) */
#define MBED_CRC_ARM_THUMB2     0
#define MBED_CRC_THUMB1         1
#else // __ARM_ARCH_xxx
#error "Unknown ARM architecture for CRC optimization"
#endif // __ARM_ARCH_xxx
#else // __arm__ || defined __ICC_ARM__ || defined __ARM_ARCH
    /* Seem to be compiling for non-ARM, or an unsupported toolchain, so stick with C implementations */
#define MBED_CRC_ARM_THUMB2     0
#define MBED_CRC_THUMB1         0
#endif
```

There should be `__ARM_ARCH` defined, it comes from CMSIS. This is related to the Mbed OS target rather than host system.

0xc0170, 2022-02-01:
You should be able to build, this is not a problem of your host system but rather the error is related to the Mbed OS target and their architecture.

You can use https://github.com/ARMmbed/mbed-tools/ to invoke CMake but will require to select your Mbed OS target. See https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/tools/cmake.md for some details.

Start with https://github.com/ARMmbed/mbed-os-example-blinky/, you can modify this example to execute the loops you want to.

0xc0170, 2022-02-02:
I am closing this as answered. If you got further questions, you can ask on our forum (But following blinky example readme should answer the questions here).

closed 2022-02-02 by 0xc0170