ARMmbed/mbed-os #15234: K22: RTC time will get reset on every bootup

Arjun765, 2022-02-17:
### Description of defect

As we can see in `targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/rtc_api.c`, RTC is only considered enabled if a `rtc_write` call has been made. This would mean i would need to set the time with `set_time` every bootup or else the first time i call `time()` RTC will be reinitialized and time will be lost. 

#### Target(s) affected by this defect ?

K22F

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.15.1

mbedmain, 2022-02-17:
@Arjun765 thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-02-21:
This line is incorrect, isn't it: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/rtc_api.c#L62 ?

It should just be checking rtc peripheral registers not the static variable . Can you send a pull request fixing the issue?

Arjun765, 2022-02-21:
Change was made in 1e46895f75a9481c812de5fbfae1a3781466291e, there probably was a reason why this was done.
@mprse 
> Can you send a pull request fixing the issue?

Does mbed still have a CLA?

0xc0170, 2022-02-21:
> Change was made in https://github.com/ARMmbed/mbed-os/commit/1e46895f75a9481c812de5fbfae1a3781466291e, there probably was a reason why this was done.

I can't see why it would be checking software value rather just reading the registers.. I checked some other implementations, they looks correct. we should probably remove the static variable there as it is not needed, is it?

>Does mbed still have a CLA?

No it does not.