ARMmbed/mbed-os #15237: USB Serial for MAX32630 and MAX32625 devices - previously working in OS6.7 not working in current versions of OS6

kcpersaud, 2022-02-23:
### Description of defect

USB Serial device support for MAX32630FTHR and MAX32625 devices was enabled in OS6.7 but seems to have disappeared in higher versions.

#### Target(s) affected by this defect ?
Maxim devices MAX32630/MAX32625

#### Toolchain(s) (name and version) displaying this defect ?
MBED Studio C6

#### What version of Mbed-os are you using (tag or sha) ?

mbedmain, 2022-02-23:
@kcpersaud thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

kcpersaud, 2022-02-23:
> The usb device support was added #14075<https://github.com/ARMmbed/mbed-os/pull/14075> . I don't recall it was removed later, can you specify what means "disappearing" ?
>
> I can see USBDEVICE present for MAX32630FTHR

Dear Martin
Thanks for your rapid reply.
I am compiling the program below with MBED Studio

The mbed_app.json
Is {
    "target_overrides": {
        "*": {
                        "target.device_has_add": ["USBDEVICE"]

        }
    }

}

#include "mbed.h"
#include "ThisThread.h"
#include "USBSerial.h"
#include "max32625pico.h"

// Virtual serial port over USB
//USBSerial microUSB;
USBSerial microUSB;

DigitalOut rLED(LED1);
DigitalOut gLED(LED2);
DigitalOut bLED(LED3);

MAX32625PICO pico(MAX32625PICO::IOH_3V3, MAX32625PICO::VIO_IOH, MAX32625PICO::VIO_IOH);

// main() runs in its own thread in the OS
// (note the calls to ThisThread::sleep_for below for delays)
int main()
{
    rLED = LED_ON;
    gLED = LED_ON;
    bLED = LED_OFF;

    rLED = LED_OFF;

    while(1) {
        microUSB.printf("micro USB serial port\r\n");

        bLED =  !bLED;
        ThisThread::sleep_for(200ms);
    }
}

With compiling in DEBUG mode and linking to MBEDOS6.7 the size of the bin file is 44,444 bytes and the program runs correctly – a CDC device is created.
When doing the same but using MBEDOS6.15.1 – the size of the bin file is 42,896 bytes and the does not create a  USB CDC device and the program does not run.

Is there some change to the json file needed?
Thanks

Professor Krishna Persaud
Department of Chemical Engineering and Analytical Science
The University of Manchester
Manchester
M13 9PL
UK.
Tel. +44(0)1613064892

0xc0170, 2022-02-25:
The implementation has not changed (usb): https://github.com/MaximIntegrated/mbed-os/blob/master/targets/TARGET_Maxim/USBPhy_Maxim.cpp 

There could be one of the changes to the target folder https://github.com/MaximIntegrated/mbed-os/commits/master/targets/TARGET_Maxim

I can't help further, the best would be  to bisect this and find out what broke the usb.

cc @ARMmbed/team-maximintegrated

ahmetalincak, 2022-02-25:
Hi @kcpersaud,

Thanks for raising the problem.

We will investigate and solve the problem as soon as possible.

Regards,

Ahmet.

kcpersaud, 2022-02-25:
Thank you very much for your help. I look forward to hearing from you.

Prof. Krishna Persaud  PhD, FRSC, FinstMC
Department of Chemical Engineering
The University of Manchester
Manchester
M13 9PL
UK
Tel. Office +44 (0) 161 306 4892
Microsoft Teams Tel No.  +441615293003

kcpersaud, 2022-02-28:
> Hey @kcpersaud<https://github.com/kcpersaud>,
>
> I found that the problem is not related to USB Device modules. MAX32630FTHR has different problem for version 6.10 and above with only ARM compilers. I haven't found the solution yet but you can work with GCC compilers while we're investigating the problem. I checked and verified that GCC_ARM works without issue.
>
> To use GCC for compiling on MBED Studio, please follow instructions here: https://os.mbed.com/docs/mbed-studio/current/installing/switching-to-gcc.html

Thank you Ahmet for your kind help in this matter.

Professor Krishna Persaud FRSC, FinstMC
Department of Chemical Engineering and Analytical Science
The University of Manchester
Manchester M13 9PL
UK
Tel. +44(0) 1613064892

0xc0170, 2022-03-01:
@kcpersaud I edited your replies. Please if you use email to reply to Github, do not add your signature as it contains information I believe should not be shared publicly.

ahmetalincak, 2022-03-14:
@kcpersaud We found the reason of this problem and trying to find a generic solution, Please consider reviewing #14465 for details.

PR #15251 (2022-03-31): Correct heap region calculation for Maxim targets

### Summary of changes 

I corrected heap region calculation for MAX32620, MAX32625 and MAX3263x. 
Fixes #15237 and #14712.

#### Impact of changes 

#### Migration actions required
closed 2022-03-31 by 0xc0170