ARMmbed/mbed-os #15241: mbed_config.h not interpreting wifi passwords inserted into mbed_app.json if they include " char in the password

Gerriko, 2022-03-06:
### Description of defect

This error relates to Wireless LAN and the WiFi example (mbed-os-snippet-TCPSocketWiFi) as found in GitHub repository: https://github.com/ARMmbed/mbed-os-snippet-TCPSocketWiFi/tree/v6.15

As per README.md it informs that we should edit mbed_app.json to include correct SSID and Password.

Now, as passwords can include any ASCII readable character, the mbed OS process to convert the password provided in mbed_app.json to what's contained in the mbed_config.h (as found in the Mbed Studio build folder) has a problem if the " character (0x22) is included in the password.

So if for example you use: my"P"assword as your password

    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"SSID\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"my\"P\"assword\""
        }
    },

The **mbed_config.h** file creates:
`#define MBED_CONF_APP_WIFI_PASSWORD       "my"P"assword"
`
but then clang misinterprets this as "my" P "assword" which creates an error in the code when used.

However, if I manually change **mbed_config.h** to
`#define MBED_CONF_APP_WIFI_PASSWORD       "my\"P\"assword"
`

clang is happy.

#### Target(s) affected by this defect ?
all targets that accept wifi API.

#### Toolchain(s) (name and version) displaying this defect ?

ARM Compiler | 6.16 | up to date |  
Clangd | 12.0.2 | up to date |  
Example projects | 1.10.1 | up to date |  
GDB client | 6-2017-q2-update | up to date |  
Git | 2.23.1 | up to date |  
Mbed Library Cache | 1.10.0 | up to date |  
Python tools | 0.6.0 | up to date |  
Debug Packs | 1.1.0 | u

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS version 6.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2022-03-06:
@Gerriko thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

pea-pod, 2022-03-24:
If you want a quotation mark to appear in your final C code, you have to work backwards from what you want to where you start. You start with the final double quote mark, get it's C equivalent, and get that intermediary's json equivalent.

We'll take your example (`my"P"assword`) as the example.

In order to get this to compile in C/C++, you need to escape the middle quotation marks. So the C version would look like this:

```C
#define MBED_CONF_APP_WIFI_PASSWORD "my\"P\"assword"
```

That gets us back to C, but you need to pop the stack one more time, since you're starting with a `.json` file. It will treat double quotes as starting and ending lines as well. So you need to make sure that you are also escaping those as well.  You did that correctly, I believe. But the final change is to make sure that the backslashes also make it into the `.h` file, so you need to escape those in the `.json` file as well.

In the relevant portion of your `mbed_app.json` file, you would have this:

```json
    "value": "\"my\\\"P\\\"assword\""
```

Because of the double conversions `\\\"` in json becomes `\"` in C, which then gets changed to the double quote in the compiled string that you're expecting. I know this because I ran into the same issue about the same time you did.

Gerriko, 2022-03-24:
Wow, thanks for that. I'm glad you figured this one out. Well done.

0xc0170, 2022-03-28:
Thank you @pea-pod for the answer.

I'll close this as resolved.

If you consider this should be covered in the docs, please add it to the https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/program-setup/advanced/configuration.md#using-configuration-data-in-code section

closed 2022-03-28 by 0xc0170