ARMmbed/mbed-os #15244: Using Break on Serial Interface is blocked by BufferedInterface

thl-mot, 2022-03-11:
### Description of defect

I need to send a break signal on serial wire. Due to thre private declaration of SerialBase on BufferedSerial the method send_break() is not usable on BufferedSerial.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.15.1

Commit: 2eb06e76208588afc6cb7580a8dd64c5429a10ce
Imported from: https://github.com/ARMmbed/mbed-os/

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed-Studio

#### How is this defect reproduced ? 

Call to method send_break() is marked as error by the compiler.

mbedmain, 2022-03-11:
@thl-mot thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-03-11:
One way would be to expose the method via BufferedSerial. You can send a pull request for review.

thl-mot, 2022-03-11:
My Hardware is a LoRa-E5  (Based on MCU_STM32WLE5xC, "device_name": "STM32WLE5JCIx").