ARMmbed/mbed-os #15245: STM32G4: I2C fast mode incorrect read

pilotak, 2022-03-11:
### Description of defect
I found that reading an I2C bus at 100kHz is fine but at 400kHz does not work correctly. It suppose to read 2 bytes. Please see below

Same code is running on STM32F303 without any problems

### 100kHz
[image]

### 400kHz
[image]

#### Target(s) affected by this defect ?
STM32G4 specifically G484VE

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

a58f8ddae773d3313921b90cd1ca2b00b6670007

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

```cpp
#include "mbed.h"
I2C i2c(I2C_SDA, I2C_SCL);
#define ADDR (0x42<<1)

char buf[2] = {0};

int main() {
    i2c.frequency(400000);

    buf[0] = 0xFD;
    i2c.write(ADDR, buf, 1, true);

    i2c.read(ADDR, buf, 2);
    printf("[0]=%X [1]=%X\n", buf[0], buf[1]);
    return 1;
}
```

jeromecoutant, 2022-03-11:
Hi Pavel

I can't understand values there:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32G4/i2c_device.h#L28-L31

Can you check with:
```
// Common settings: I2C clock = 160 MHz, Analog filter = ON, Digital filter coefficient = 0
#define TIMING_VAL_160M_CLK_100KHZ  0x90A13E56  // Standard mode with Rise Time = 400ns and Fall Time = 100ns
#define TIMING_VAL_160M_CLK_400KHZ  0x30D2153A  // Fast mode with Rise Time = 250ns and Fall Time = 100ns
#define TIMING_VAL_160M_CLK_1MHZ    0x10830F28  // Fast mode Plus with Rise Time = 60ns and Fall Time = 100ns

```

Jerome

pilotak, 2022-03-11:
https://github.com/ARMmbed/mbed-os/blob/170aea5932692f906030e77eaae98bad96f6d10c/targets/TARGET_STM/TARGET_STM32G4/i2c_device.c#L29
gives me `160000000`

https://github.com/ARMmbed/mbed-os/blob/170aea5932692f906030e77eaae98bad96f6d10c/targets/TARGET_STM/i2c_api.c#L732

is actually called 4 times *(i can undestand 2 tries as first call is in constructor and second my set to 400kHz)*
```
timing: C0311319
timing: 10B1102E
timing: 10B1102E
timing: 10B1102E
```

jeromecoutant, 2022-03-15:
But values are wrong, try to updates values with the one in my previous comment

pilotak, 2022-03-16:
Oh sorry, my misunderstanding. With your values it works for both 100kHz and 400kHz

PR #15249 (2022-03-18): STM32G4: Fix I2C timing

### Summary of changes 

Fixes wrong I2C timing on STM32G4 @ 160MHz

Fixes #15245
#### Impact of changes 

#### Migration actions required
closed 2022-03-18 by 0xc0170