ARMmbed/mbed-os #15250: Support for Nucleo U575ZI-Q

fliescher, 2022-03-19:
### Description of defect
Please add support for the Nucleo U575ZI-Q Board. I'm able to program and flash with Mbed Studio. However debugging doesn't work as it's officially not yet supported.

#### Target(s) affected by this defect ?
Nucleo-U575ZI-Q

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2022-03-19:
@fliescher thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2022-03-21:
Hi

NUCLEO_U575ZI is already supported with mbed-os.

https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32U5/TARGET_STM32U575xI/TARGET_NUCLEO_U575ZI_Q

https://os.mbed.com/platforms/NUCLEO-U575ZI-Q/

If you have trouble with Mbed Studio, please maybe check https://os.mbed.com/docs/mbed-studio/current/targets/custom-targets.html

closed 2022-03-21 by fliescher

daniel-merrick, 2022-03-21:
I am also having problems with debugging the NUCLEO-U575ZI-Q board.

I can successfully compile and flash the board, however the DEBUG symbol is grayed out and can't be selected. Are there any expectations to support this?

I am using the latest master branch and mbed version 1.10.4

daniel-merrick, 2022-03-21:
@fliescher Were you able to get the debugger in MBED Studio to work?

fliescher, 2022-03-21:
@daniel-merrick 
In the target selection hit the "manage custom targets" button (right next to the search line). 
Selecting a Deploy and debug target turns the Debug button active (I tested it with the cortex-m option). However I wasn't able to test it in depth yet due to lag of time

daniel-merrick, 2022-03-21:
@fliescher Thanks for the response.

When I select manage custom targets this is my configuration:
USB Device: STLINK V3
Target Name: NUCLEO-U575ZI-Q
Build Target:  NUCLEO-U575ZI-Q

However when I go to Deploy and Debug target, I don't have any option to select 'cortex-m'.

Did you have to configure anything else to select 'cortex-m' ?

fliescher, 2022-03-21:
sorry, the exact name is cortex_m. After selecting it, it's shown as 'cortex_m (Native)'

But please notice: I didn't test it in debug mode yet. But as the U575 holds an Arm Cortex-M33 core this selection makes sence in my opinion

daniel-merrick, 2022-03-22:
Actually I re-installed mbed-studio now I can select 'cortex_m'. Seems like it wasn't loading the PyOCD options for some reason. 

Thanks!