ARMmbed/mbed-os #15252: CAN FORMAT problems

Lynda96, 2022-03-22:
Hi
I am working on CAN bus with NUCLEO BOARD ,so I am coding with mbed studio , I am just doing a loop test (send data from CAN1TD and receive it from CAN2RD with the same board ) this is the code that I have tested
[image]

it doesn't work with EXtended format ,it works only with Standard format 
Can you please ,tell me where is the problem ? 
Thank you

mbedmain, 2022-03-22:
@Lynda96 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-03-31:
cc @ARMmbed/team-st-mcd

pilotak, 2022-04-01:
I have never tried this direct wiring, only with transceivers to get CAN but in my code i also set
`messageOut.type = CANData;`

jeromecoutant, 2022-04-19:
Hi @Lynda96 
Will you be able to push a patch for that ?
Thx

Lynda96, 2022-04-20:
Hi @jeromecoutant 
I don't understand what do you mean ?

jeromecoutant, 2022-04-20:
> obj->CanHandle.Init.ExtFiltersNbr  was equal to 128 I changed it to 64 and it works

Do you think you can propose a clean patch that is working for every one ?

Lynda96, 2022-04-20:
it works for me but i don't know if it's gonna work for you ,i change it cause on datasheet Can EXTENDED Filter is 64 elements ,so i think it's depends to the board you are working on .
I am working on NucleoBoard h743zi2
[image]

JohnK1987, 2022-10-08:
The basic rules were not followed here

> Could you add some more detail to the description? A good description should be at least 25 words.
> What target(s) are you using?
> What toolchain(s) are you using?
> What Mbed OS version are you using? [...]

@Lynda96 did not fill all necessary information for full context, especially what MbedOS version was used.

This issue is know issue and it was already solved in MbedOS 6.14 https://github.com/ARMmbed/mbed-os/pull/15000
This one can be closed. What do you think @0xc0170 @jeromecoutant ?

closed 2023-01-05 by 0xc0170