ARMmbed/mbed-os #15253: Wrong MBED_ROM_START adress for Portenta H7

Flydroid, 2022-03-23:
### Description of defect

Compiling code for the Portenta H7 with the Arduino bootloader expects the code to start at mbed_rom_start=0x08040000.
The current targets.json this value is set to 0x08000000
https://github.com/ARMmbed/mbed-os/blob/99cb88541a054ff1c356d36594fe08dfe1eede75/targets/targets.json#L3481

#### Target(s) affected by this defect ?

Portenta H7 M7

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 6.9.0
-->

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Platformio 
platformio-stm32 @ 15.2.0
framework-mbed @ 6.60900.210318 (6.9.0)

#### How is this defect reproduced ? 

flashing the blinky script will not not run with out mbed_rom_start=0x08040000 changed in target.json in the framework-mbed folder in platformio/packages

mbedmain, 2022-03-23:
@Flydroid thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Flydroid, 2022-03-23:
@mbedmain I have updated the issue description

jeromecoutant, 2022-03-24:
@facchinm

facchinm, 2022-03-24:
Hi @Flydroid ,
we decided to keep the "project" start at the beginning of the flash for the following use cases:
* users that don't need a bootloader
* users that want to write a bootloader

Changing that filed in your own `mbed_app.json` is trivial https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/conf/mbed_app.json#L17 so I think that right way to handle this is by writing some better documentation :wink: 
@sebromero

Flydroid, 2022-03-24:
Alright, the mbed_app.json config works!
Better documentation is always appriciated!

I will also open an issue with platformio so the mbed framework gets added to the portenta H7.

closed 2022-03-24 by Flydroid

sebromero, 2022-03-25:
@facchinm @Flydroid I added the information about where to adjust the start address here: https://github.com/arduino/docs-content/pull/119