ARMmbed/mbed-os #15254: mbed_override_console with USBSerial doesn't work on Portenta H7

Flydroid, 2022-03-25:
### Description of defect

Using the mbed_override_console to have printf output via USBSerial doesn't work on the Portenta H7.
The implementation has followed the [[Hitchhiker’s Guide to Printf in Mbed 6](https://forums.mbed.com/t/hitchhikers-guide-to-printf-in-mbed-6/12492)]( https://forums.mbed.com/t/hitchhikers-guide-to-printf-in-mbed-6/12492)
and 

#### Target(s) affected by this defect ?
Portenta H7

#### Toolchain(s) (name and version) displaying this defect ?
gccarmnoneeabi@1.50401.211104
through platformio

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 6.9.0
"bare-metal" profile with "requires": ["bare-metal","drivers-usb","events"],

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Platformio
platformio-stm32 @ 15.2.0
framework-mbed @ 6.60900.210318 (6.9.0)

#### How is this defect reproduced ? 
https://gist.github.com/Flydroid/85b9868c1226e0ad69572fb2740465c8
Executing the code on the Portenta H7 gives a blinking blue LED and "Hello World" on the COM port.
The printf statement "Hello World2" is not visiable on the COM port.
USBSerial is started in non-blocking mode.

I have tried any combinatation of mbed_override_console and mbed_target_override_console

0xc0170, 2022-03-28:
cc @facchinm

Flydroid, 2022-03-28:
Update:
HellWorld2 is actually beeing set but as a block of ~85 prints every 45 seconds.
USBSerial::sync() didn't do any effect.

Flydroid, 2022-03-28:
fixed!

Termination charater was missing in printf statement

closed 2022-03-28 by Flydroid