ARMmbed/mbed-os #15257: NUCLEO_G031k8 I2C_2 ERROR

ObeidHaidar, 2022-04-01:
I keep getting the following error when trying to use I2C2 pins A4(PA_12) and A5(PA_11)
```
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x80038F3
Error Value: 0x0
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_G031K8
-- MbedOS Error Info --
I2C: unknown instance
```
Code works fine when I use any of the other SPI1 pins. 

Checked the PeripheralPins.c file for the board and it seems fine. 

example of code that would fail:
```
#include "I2CSlave.h"
#include "I2C.h"
#include <cstdio>
#include <string.h>
#define SLAVE_ADDR 0xA0
#define BUFFER_SIZE 6

I2C master(A4,A5);

static const char* to_send[] = { "abcde", "12345", "EFGHI" };

int main() {
    char buf[BUFFER_SIZE];
    int send_index = 0;
    while (1) {
        strcpy(buf, to_send[send_index]);
        // Write the new message to the slave
        if (master.write(SLAVE_ADDR, buf, BUFFER_SIZE)) {
            printf("Failed to write to slave!\n");
        } else {
            printf("Written to slave: %s\n", buf);
        }
        // Read what the slave has (should be identical)
        if (master.read(SLAVE_ADDR, buf, BUFFER_SIZE)) {
            printf("Failed to read from slave!\n");
        } else {
            printf("Read from slave: %s\n", buf);
        }
        // Change the message we're writing to the slave
        send_index++;
        if (send_index > 2) {
            send_index = 0;
        }
        wait_us(500000); // Wait 0.5s
    }
}
```
Can anyone try this and see if they get the same result?

_Originally posted by @ObeidHaidar in https://github.com/ARMmbed/mbed-os/issues/13006#issuecomment-1085441824_

mbedmain, 2022-04-01:
@ObeidHaidar thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

ObeidHaidar, 2022-04-01:
Target is NUCLEO_G031k8
Using mbed-os 6.13.0 via Mbed Studio

jeromecoutant, 2022-04-01:
Hi
Issue is corrected since #15075
Please use latest version of mbed-os
Regards,