ARMmbed/mbed-os #15259: Compiling too slow

WillieVisagie, 2022-04-07:
### Description of defect
    Compiling the project are simply too slow.  It's literally a show stopper.  What can be done about it?  This is such an awesome platform but the compile time is killing it.

#### Target(s) affected by this defect ?
    All targets

#### Toolchain(s) (name and version) displaying this defect ?
    ARM GCC

#### What version of Mbed-os are you using (tag or sha) ?
    latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
    mbed-cli, GCC arm toolchain 10 2021.10, etc.

#### How is this defect reproduced ? 
    Build your project from scratch

0xc0170, 2022-04-07:
you are using `mbed build` command? What examples/tests are you testing with and what are the times you get ?

0xc0170, 2022-04-07:
Can you try to set `MBED_BUILD_TIMESTAMP` via command line:

`-DMBED_BUILD_TIMESTAMP=100` ? And enable ccache - this should make it much faster (see https://github.com/ARMmbed/mbed-os/pull/8359)

WillieVisagie, 2022-04-07:
Hi

Yes - I'm using `mbed compile --profile=debug`.

Do you mean to pass these two arguments (`-DMBED_BUILD_TIMESTAMP=100` and `ccache`) on the command line when calling `mbed compile`?

WillieVisagie, 2022-04-10:
Can anyone answer my question, please?

llawrence12, 2022-04-11:
Personally I use the .mbedignore file to remove any unneeded/unwanted things from mbed-os and that significantly drops my compile time (once in the root of the project it must be named ".mbedignore"). Maybe there's a better way to do it but I don't know/care about it since this works for me. I also don't use anything newer than os 6.8.0 with CLI 1 so things may have changed in newer versions

Here is the one I use, you can change it to include/exclude what you like
[mbedignore.txt](https://github.com/ARMmbed/mbed-os/files/8468167/mbedignore.txt)

WillieVisagie, 2022-04-12:
Great idea.  I will definitely do that.

Another idea is to use mbed-tools, or CLI version 2.  But then again, I have such trouble with the `CMakeLists.txt` files, because every time you add the `mbed-os' library, it gets recompiled.  So I guess I just need to learn `cmake` and its `CMakeLists.txt` files.

ladislas, 2022-04-13:
This problem is well known and due to the use of interface in cmake instead of static or object libs.

see https://github.com/ARMmbed/mbed-os/issues/13981#issuecomment-1003386596 and subsequent comments

there was some work done to fix that, but the changes are huge and development has slowed down.

there is a work in progress from @multiplemonomials you can find here: https://github.com/multiplemonomials/mbed-os/tree/mbed-ce-build-system-refactor

and also ongoing discussions about making a community fork of mbed-os to fix this kind of issues early and then back port them if possible to the main mbed-os repo.

rvasquez6089, 2022-04-26:
Thats pretty normal. 
Windows is the worst offender. If i need to use windows, I install Ubuntu 20.04 in WSL and compile my project in the WSL filesystem on Ubuntu. Its about 5 times faster than windows.
If I need super fast build times, I go to native Linux. its about 7 times faster than windows good luck!