ARMmbed/mbed-os #15260: Exporting to uVision MDK

WillieVisagie, 2022-04-07:
### Description of defect
When exporting to uVision6, it does not complete compile.  I exported using `mbed-cli`.  Here is the commands:
`mbed new Test`
Create empty main.cpp with `int main(void)` function.
`mbed target NUCLEO_F429ZI`
`mbed export -i uvision6`
Then open the project in Keil Vision.

It compiles up to the almost very end and then this error:

[image]

So is the problem with the exporter or where?

#### Target(s) affected by this defect ?
My target is set to NUCLEO_F429ZI

#### Toolchain(s) (name and version) displaying this defect ?
uVision: MDK-ARM Essential Version 5.36.0.0
ARM compiler V6.16

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
The `latest` tag, currently 2eb06e76208588afc6cb7580a8dd64c5429a10ce

#### How is this defect reproduced ? 
Just new and export a project with the chosen target.

mbedmain, 2022-04-07:
@WillieVisagie thank you for raising this issue.Please take a look at the following comments:

What Mbed OS version are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

WillieVisagie, 2022-04-07:
mbed version 6.15.1

0xc0170, 2022-04-07:
There was this already reported https://github.com/ARMmbed/mbed-os/issues/13166.
The exporters were deprecated long time ago, I will close this issue as "won't fix". Thanks for understanding

Look at the fault handling and find out why the symbol is missing (I believe there will be config not properly propagated and the symbol won't be part of the compilation).

closed 2022-04-07 by 0xc0170