ARMmbed/mbed-os #15264: Modification of the AT_CellularDevice constructor to customize the ATHandler delimiter

Tobi15, 2022-04-13:
### Description of defect
When you try to modify the delimitor "\r" of the ATHandler in the AT_CellularDevice, it is not possible at runtime. 

I propose to modify the constructor of AT_CellularDevice as follow:

AT_CellularDevice.h :
```
AT_CellularDevice(FileHandle *fh, char* delim = "\r");
```

AT_CellularDevice.cpp
```
AT_CellularDevice::AT_CellularDevice(FileHandle *fh, char* delim) :
    CellularDevice(),
    _at(fh, _queue, DEFAULT_AT_TIMEOUT, delim),
#if MBED_CONF_CELLULAR_USE_SMS
    _sms(0),
#endif // MBED_CONF_CELLULAR_USE_SMS
    _network(0),
    _information(0),
    _context_list(0),
    _default_timeout(DEFAULT_AT_TIMEOUT),
    _modem_debug_on(false),
    _property_array(NULL)
{
    MBED_ASSERT(fh);
}
```

#### Target(s) affected by this defect ?
All targets

#### Toolchain(s) (name and version) displaying this defect ?
All toolchains

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1

#### How is this defect reproduced ? 
When you try to modify the delimitor of the ATHandler in the AT_CellularDevice, it is not possible at runtime.

0xc0170, 2022-04-19:
I am not familiar with cellular network. First question, why would you change the delimiter (wondering why it was hardcoded to be always the same) ?

Tobi15, 2022-04-20:
So, in my case, I implemented a new driver for the nrf9160 which was not supported by mbed. This nrf9160 modem needs a specific delimiter "\n\r" to understand mbed AT commands. This is the reason why it seems interesting not to hardcode this delimiter.

0xc0170, 2022-04-20:
OK, you can create a new pull request fixing this? You can use default parameter to have this non breaking for current users (\r would be used).