ARMmbed/mbed-os #15265: STM32G4: UART5 IRQ not working

pilotak, 2022-04-13:
### Description of defect

When using `UnbufferedSerial` on UART5, irq is never called, so i looked at the https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32G4/serial_device.c and i discovered that there is no UART5 support so i simply added following, but still no success. Reading the serial "classic" way in `main()` works. 

On USART1 irq works as expected. 

> During search i also found that `UART4_BASE` is incorrectly set as `USART4_BASE` and `USART3_4_LPUART1_IRQn` doesn't exists it should be `UART4_IRQn`. Also under LPUART irq is set to `USART3_4_LPUART1_IRQn` but should be `LPUART1_IRQn` but that is not part of this issue.

to the top
```cpp
#if defined(UART5_BASE)
static void uart5_irq(void)
{
    uart_irq(UART_5);
}
#endif
```

in `serial_irq_set()`
```cpp
#if defined(UART5_BASE)
    if (obj_s->uart == UART_5) {
        irq_n = UART5_IRQn;
        vector = (uint32_t)&uart5_irq;
    }
#endif
```

and to `serial_get_irq_n()`
```cpp
#if defined(UART5_BASE)
        case UART_5:
            irq_n = UART5_IRQn;
            break;
#endif
```

Is there anything else that needs to be done in order to get irq on UART5 working?

#### Target(s) affected by this defect ?
STM32G4 specifically G484VE

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

99cb88541a054ff1c356d36594fe08dfe1eede75

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

```cpp
#include "mbed.h"

UnbufferedSerial serial(PC_12, PD_2); // UART5
//UnbufferedSerial serial(PA_9, PA_10); // USART1

volatile bool test = false;

void cb() { // ISR
    char rec = 0;
    serial.read(&rec, 1); // clear

    test = true;
}

int main() {
    serial.baud(115200);
    serial.format(8, SerialBase::None, 1);
    serial.attach(callback(cb));

    while (1) {
        /*while (serial.readable()) {
            char rec = 0;

            serial.read(&rec, 1);
            printf("data: %02X\n", rec);

        }*/
        if (test) {
            test = false;
            printf("test\n");
        }
    }
}
```

0xc0170, 2022-04-19:
cc @ARMmbed/team-st-mcd

pilotak, 2022-04-20:
Many thanks @jeromecoutant  for stepping into this, the PR changes are exacly the same i did last week locally, but unfortunately UART5 IRQ is not working

jeromecoutant, 2022-04-20:
> the PR changes are exacly the same i did last week locally, but unfortunately UART5 IRQ is not working

All changes....? :-(

jeromecoutant, 2022-04-25:
Could you check with?
`__HAL_RCC_UART5_CONFIG(RCC_UART5CLKSOURCE_HSI);`

pilotak, 2022-04-26:
I changed it at https://github.com/ARMmbed/mbed-os/blob/decc6d022cb10f74d31410d49e03c571c992f70c/targets/TARGET_STM/TARGET_STM32G4/STM32Cube_FW/STM32G4xx_HAL_Driver/stm32g4xx_hal_rcc_ex.c#L273 and it works, strangely when putting it back it it works too, maybe it gets remebered...

jeromecoutant, 2022-04-26:
I don't think that current code is calling the part you mentioned.
And maybe "HSI clock on" is mising in my proposition....
But I changed #15269 yesterday, could you check and maybe approve ? :-)

pilotak, 2022-04-26:
Silly me, i have checked out the yesterday PR and also did HSI change in one go, that is the reason

PR #15269 (2022-04-26): STM32G4 : enable UART ASYNC
### Summary of changes 

Fix #15265

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2022-04-26 by 0xc0170