ARMmbed/mbed-os #15266: Nucleo-L073RZ I2C2 not defined - Unknown instance

Mircerson, 2022-04-14:
### Description of defect

#### Target(s) affected by this defect ?
STM32L073RZTx
Nucleo-L073Rz

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?

mbedmain, 2022-04-14:
@Mircerson thank you for raising this issue.Please take a look at the following comments:

What target(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Mircerson, 2022-04-14:
Target currently used is a Nucleo-L073RZ and a custom board built around the STM32L073. The bug is present on both boards.
The mbed version is 6.15.1.
Using mbed studio ( latest ) and GCC.
See above to reproduce.

0xc0170, 2022-04-19:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2022-04-19:
> Checking i2c_device.c, there is no I2C_2.

Good catch :-(
I am correcting it asap.

jeromecoutant, 2022-04-19:
@Mircerson please check #15268

Mircerson, 2022-04-19:
Cheers, I'll give it a go tonight

Mircerson, 2022-04-21:
@jeromecoutant  thanks for the fix, all working as intended now 👍

PR #15268 (2022-04-26): STM32L0 : I2C2 was missing
### Summary of changes 

Fix #15266 

After #14557, I2C2 was missing for STM32L0 in the clock source configuration.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2022-04-21 by Mircerson