ARMmbed/mbed-os #15278: MbedTLS included to Mbed OS has open security issues

JanneKiiskila, 2022-05-02:
### Description of defect

https://github.com/ARMmbed/mbed-os/blob/master/connectivity/mbedtls/include/mbedtls/version.h

Mbed TLS included to Mbed OS has open security issues, see Mbed TLS itself.
https://tls.mbed.org/tech-updates/security-advisories
Based on information there one should use at least 2.28.0.

I find it a bit problematic that Mbed TLS is actually included to Mbed OS itself, it does not encourage easy/quick updates of Mbed TLS as you have to get the full Mbed OS release to get it updated.

In minimum Mbed OS team should provide information how to update Mbed TLS standalone.
- Is it enough to add the mbedTLS folder to an .mbedignore file? Probably a bit different with Mbed OS 6 and cmake based tooling.

Or should you do minor Mbed OS release updates anytime a security vulnerability is found on Mbed TLS?

How should mbed-os 5.15 branch be handled? In my humble opinion 2.28.0 should be updated there as well, so new Mbed OS 5.15.x release would be needed as well.

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

Master, 9f326aa8b279f9e162588c4991ab10af6d28e2f2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Compile anything using MbedTLS.

0xc0170, 2022-05-09:
> In minimum Mbed OS team should provide information how to update Mbed TLS standalone.

To import mbed tls, the importer script is used: https://github.com/ARMmbed/mbed-os/blob/master/connectivity/mbedtls/tools/importer/Makefile#L21 (referencing the info how to use it). You can run it to reimport the newer version.

JanneKiiskila, 2022-05-09:
@0xc0170 - why are we not doing 5.15.9 already then?-)