ARMmbed/mbed-os #15283: STM32WL I2C not working in Release, no issue in Debug/Develop

hallard, 2022-05-07:
### Description of defect

Working with ATECC I2C device for LoRa, in `Debug` or `Develop` mode works like a charm, as soon as I build in `Release` mode, device is not seen anymore and can't talk with it. Everything else is working fine (LoRaWAN and app job)

Tried basic I2C scanner, same thing, device is not seen building `Release`

#### Target(s) affected by this defect ?

STM32WL LoRa-E5 (may be other)

#### Toolchain(s) (name and version) displaying this defect ?

[image]

#### What version of Mbed-os are you using (tag or sha) ?

6.14
6.15.1
master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio

#### How is this defect reproduced ? 

Switch Build profile to 'Debug', 'Release' or 'Develop'

jeromecoutant, 2022-05-09:
Hi
Only 2 difference between release and develop profile...
- NDEBUG (only in release)
- MBED_TRAP_ERRORS_ENABLED=1 (only in develop)

hallard, 2022-05-09:
I just split off I2C stuff onto a basic program just to be sure, can't be easier, in the same time I wired a 2nd IC2 device on same bus (9 DoF sensor)

```cpp
#include "mbed.h"

I2C i2c(PA_15, PB_15); 

int main(void)
{
	printf("\n***** I2C Scanner *****\n");
        ThisThread::sleep_for(2s);

	int count = 0;
	for (int address = 0; address < 127; address++)	{
		if (!i2c.write(address << 1, NULL, 0)) {																								  
			printf("I2C device found at address 0x%02x (8-bits 0x%02X)\n", address, address << 1);
			count++;
		}
        ThisThread::sleep_for(20ms);
	}
	printf("%d device(s) found\n", count);
}
```

`debug` output
````
***** I2C Scanner *****
I2C device found at address 0x58 (8-bits 0xB0)
I2C device found at address 0x69 (8-bits 0xD2)
2 device(s) found
````

`develop` output
````
***** I2C Scanner *****
I2C device found at address 0x58 (8-bits 0xB0)
I2C device found at address 0x69 (8-bits 0xD2)
2 device(s) found
````

`release` output
````
***** I2C Scanner *****
I2C device found at address 0x69 (8-bits 0xD2)
1 device(s) found
````

and as final test just to be sure, I switched 9Dof Sensor with a 2nd ATECC608 (adafruit stema board)

`Develop` output
````
***** I2C Scanner *****
I2C device found at address 0x58 (8-bits 0xB0)
I2C device found at address 0x60 (8-bits 0xC0)
2 device(s) found
````

`Release` output
````
***** I2C Scanner *****
0 device(s) found
````

Not sure what's going on on I2C stuff in release mode, looks like ATECC does not answer in release, would love to know results of this piece of code if anyone has a board with ATECC chip

jeromecoutant, 2022-05-09:
Does it change something if i2c is not global ?

hallard, 2022-05-11:
Ok got some investigations and found the root cause, it's ATECC Secure Element, I discovered sometime I need to scan I2C bus with address before to get the device seen. So I tried to detect the device several time to see it.

```cpp
#include "mbed.h"
I2C i2c(PA_15, PB_15); // SDA, SCL
int main(void)
{
    int i, address;
    printf("\n***** I2C Scanner *****\n");
    i2c.frequency(100000);
    ThisThread::sleep_for(1s);
    
    address = 0x58;																						     
    printf("I2C Check address 0x%02x (8-bits 0x%02X) => ", address, address << 1);
    for (i=0; i<8; i++) {
        if (!i2c.write(address << 1, NULL, 0)) {
            printf("Found after %d queries\n", i+1);
            return 0;
        }
        ThisThread::sleep_for(2ms);
    }
    printf("Not found after %d queries\n", i+1);
}
```

When running in `Develop` or `Debug` sometime it is seen after 2 queries other 6 or 8. 

And reading around seen some send 4 x 00 on I2C bus at 100KHz to wakeup the device (this pull SDA low enough time to wake the device), Yes, because it need to be waked pulling SDA line low for sometime.

So I dirty fixed like that and no more problem 

````cpp
#include "mbed.h"
I2C i2c(PA_15, PB_15); // SDA, SCL
int main(void)
{
    int i, address;
    printf("\n***** I2C Scanner *****\n");
    i2c.frequency(100000);
    ThisThread::sleep_for(1s);
    // Force ATECC to wake up then deinstantiate  DigitalOut
    {
        DigitalOut sda(PA_15);
        sda = 0;
        ThisThread::sleep_for(10ms);
        sda = 1;
        ThisThread::sleep_for(2ms); // Let time to wake up
    }

    address = 0x58;																						 
    printf("I2C Check address 0x%02x (8-bits 0x%02X) => ", address, address << 1);
    if (!i2c.write(address << 1, NULL, 0)) {
        printf("Found\n");
    } else 
        printf("Not Found\n");
    }
}
````

With this dirty fix it works in `Release` and all mode. But the deal is that using [mbed-CryptoAuthLib](https://github.com/ARMmbed/mbed-cryptoauthlib) and  [CryptoAuthLib](https://github.com/ARMmbed/cryptoauthlib) has the same issue, so in my code I needed to wake up every time I call crypto function that call `atcab_init()`(because I like to put device into sleep after use calling `atcab_release();`

So I don't know who's guru in charge of that but this means official examples [mbed-os-example-atecc608a](https://github.com/ARMmbed/mbed-os-example-atecc608a) does not works either in release. I will open issue on this one.