ARMmbed/mbed-os #15285: STM32G4 UART5 reading at low speed not working

pilotak, 2022-05-11:
Reading an UART5 either in loop or via IRQ is not working at 2400 or 1200, with 4800 and above it works fine.
The result is always `0x00, 0x00, 0x00` per character regardless what i send from from other device

### Description of defect

#### Target(s) affected by this defect ?
STM32G4 specifically G484VE

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

88b6bb0d8568d0249b6a072b5e5772fb159dafa6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

```cpp
UnbufferedSerial serial(NC, PD_2); // UART5

int main() {
    serial.baud(2400);

    while (1) {
        while (serial.readable()) {
            char rec = 0;

            serial.read(&rec, 1);
            printf("data: %02X\n", rec);
        }
    }
}
```

mbedmain, 2022-05-11:
@pilotak thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2022-05-11:
Hi
I think we need to check this part of MCU reference manual:
37.5.9 USART Auto baud rate detection
The clock source frequency must be compatible with the expected communication speed.
• When oversampling by 16, the baud rate ranges from usart_ker_ck_pres/65535 and usart_ker_ck_pres/16.
• When oversampling by 8, the baud rate ranges from usart_ker_ck_pres/65535 and usart_ker_ck_pres/8.

pilotak, 2022-05-12:
Looking at the current implementation "oversampling by 16" is set by default so minimum speed is 2441 so a quick fix is to change prescaler to 16 which would cover the low speed stuff down to 153.

https://github.com/ARMmbed/mbed-os/blob/e2e927e2ee0742e6d360870b05f6d31bd0517582/targets/TARGET_STM/serial_api.c#L633
```cpp
#if defined(TARGET_STM32G4)
    if (obj_s->baudrate < 4800) {
        huart->Init.ClockPrescaler = UART_PRESCALER_DIV16;
    } else {
        huart->Init.ClockPrescaler = UART_PRESCALER_DIV1;
    }
#else
    huart->Init.ClockPrescaler = UART_PRESCALER_DIV1;
#endif // TARGET_STM32G4
```

jeromecoutant, 2022-05-12:
I think you can even propose in a PR something like:
````
#if defined(UART_PRESCALER_DIV1) // G0/H7/L4/L5/U5/WB/WL
    if (obj_s->baudrate < 4800) {
        huart->Init.ClockPrescaler = UART_PRESCALER_DIV16;
    } else {
        huart->Init.ClockPrescaler = UART_PRESCALER_DIV1;
    }
#endif
````

pilotak, 2022-05-13:
@jeromecoutant thanks a lot!

PR #15286 (2022-05-13): STM32G4: Fix serial port at low speed baud

### Summary of changes 
Fixes serial port at low speed (< 4800) on STM32G4 family

Fixes #15285

#### Impact of changes 

#### Migration actions required
closed 2022-05-13 by 0xc0170