ARMmbed/mbed-os #15292: multiple SPI objects cause conflicts on NUCLEO-L432KC

Flydroid, 2022-06-01:
### Description of defect

I have a [NUCLEO-L432KC](https://www.st.com/en/evaluation-tools/nucleo-l432kc.html) a where two SPI channels are are used.
Nothing is connected except a Ikalogic SP209 logic analyzer.

| Pin      | spiDevice1 | spiDevice2 |
|--------|------------|--------------|
| MOSI  |  PB_5        | PA_7          | 
| MISO  | PB_4         | PA_6          | 
| SCLK   | PB_3         | PA_5          | 
| CS       | PA_4         | PB_0          | 

The solder bridges SB16 and SB18 on PA_6 and PA_5 have been removed. 

The SPI buses work as expected (data on the bus) when only one of them is used (spiDevice.write(0xFF) )in the program (spiDevice1 and spiDevice2 ). Both buses are defined in the beginning of of the program, but this doesn't seem to influence the behaviour until actually both spiDevices are used.

When both buses are used (alternating between spiDevice1 and spiDevice2) data and clock appears on both buses at the same time- even while the other bus is not currently used (as you can see by the activation of the cs pin). First spiDevice1 is active but we also get data on MOSI and SCLK on the pins of spiDevice2 as well as vice versa when spiDevice2 is active. 

[image]

According to the STM32L432 the pins of spiDevice1 can be accessed by either SPI1 or SPI3 , whereas the pins of spiDevice2 can only be accessed by SPI1.

I have looked in to the mbeds SPI.cpp and read  the comments on issue https://github.com/ARMmbed/mbed-os/issues/10827 and https://github.com/ARMmbed/mbed-os/issues/7358 with no further inside than that it should work.  `DEVICE_SPI_COUNT ` is defined for TARGET_STM32L4 as 3.

The only explanation i can see so far is that mbed is using SP1 for both spiDevice1 and spiDevice2 somehow which results in the same data and clock appearing on both sets of pins. 

#### Target(s) affected by this defect ?

NUCLEO-L432KC

#### Toolchain(s) (name and version) displaying this defect ?

gccarmnoneeabi@1.50401.211104
through platformio

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.9.0
"bare-metal" profile with "requires": ["bare-metal","drivers-usb","events"],

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

platformio-stm32 @ 15.2.0
framework-mbed @ 6.60900.210318 (6.9.0)

#### How is this defect reproduced ? 

Run https://gist.github.com/Flydroid/08b69ad1fdd0c0360e5027b87e96b993 on NUCLEO-L432KC for alternating buses.
Comment out one section of spiDevice1 or spiDevice2 to just test one bus.

0xc0170, 2022-06-01:
cc @ARMmbed/team-st-mcd

JohnK1987, 2022-06-03:
Hello,

according to [PeripheralPins.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L432xC/TARGET_NUCLEO_L432KC/PeripheralPins.c#L161) you need to use _ALT0 sufix for alternate function of same pin
<html>
<body>

Pin | spiDevice1 | spiDevice2
-- | -- | --
MOSI | PB_5_ALT0 | PA_7
MISO | PB_4_ALT0 | PA_6
SCLK | PB_3_ALT0 | PA_5
CS | PA_4_ALT0 | PB_0

</body>
</html>

BR, Jan

Flydroid, 2022-06-03:
Hi JohnK1987,

thanks a lot! That fixed it!

Best regards,
Jonathan

closed 2022-06-03 by Flydroid