ARMmbed/mbed-os #15300: Lora E5 support for non volatile storage

jwktje, 2022-06-20:
Hello ARMmbed,

I’m not sure this is the right place to ask this question but I don’t know where to start otherwise.
So if anyone can guide me to the right resource, that would be great.

I’m using Mbed OS on the Lora-E5 from Seeed. All works great but I can’t get non volatile memory to work.
I want to store 1 integer variable even after a power-cycle.

I tried adapting the KVstore examples into my code. And they work as long as the power is on. But after a power cycle the data is gone.

Below are the two functions I’m testing with.
```
void kv_store_get(char* key) {
    /* Start by getting key's information */
    res = kv_get_info(key, &info);
    char *kv_first_value_out = new char[info.size + 1];
    memset(kv_first_value_out, 0, info.size + 1);
    res = kv_get(key, kv_first_value_out, info.size, &actual_size);
    printf("kv_get -> %d\n", err_code(res));
    printf("kv_get key: %s\n", key);
    printf("kv_get value: %s\n", kv_first_value_out);
    delete[] kv_first_value_out;
}

void kv_store_set(char* key, char* value) {
    res = kv_set(key, value, strlen(value), 0);
    printf("kv_set -> %d\n", err_code(res));
}
```
And I’ve set the following configurations in mbed_app.json:
```
"LORA_E5_MINI": {
      "stm32wl-lora-driver.rf_switch_config": 2,
      "lora.duty-cycle-on": false,
      "target.features_add": ["STORAGE"],
      "target.components_add": ["FLASHIAP"],
      "storage.storage_type": "TDB_INTERNAL",
      "storage_tdb_internal.internal_size": "(4*1024)"
  }
```
If anyone can point me into a direction of getting this to work that would be very much appreciated.
Thanks in advance

mbedmain, 2022-06-20:
@jwktje thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-06-21:
I am not that familiar with kv-store, but I see you enabled flashiap and set `TDB_INTERNAL`. Can you check if you are writing the data to the internal flash (check the address range and read back data) ? Then check what region are data being written and how it is defined in linker script (is anyone overwriting the data?) ?

jwktje, 2022-06-21:
@0xc0170 Is there another method I can/should use that you are more familiar with? I'm not committed to using kv-store. I just want to have any means of storing a single variable that is persisted after a power cycle.

0xc0170, 2022-07-04:
I'll close this as not technical issue.

closed 2022-07-04 by 0xc0170