ARMmbed/mbed-os #15301: fatal error: Python.h: No such file or directory

sw-dev-code, 2022-06-20:
I'm using the Ubuntu 20.04.4 OS with installed Python 3.7.3. When I try to use markupsafe module the following error occurs.

I also tried with the default installed Python 3.8.10 but the same error occurred. I've reached the support for both markupsafe and Python but there just pass my issue to each other.

tools.utils.ToolException: ./venv/lib/python3.7/site-packages/markupsafe/_speedups.c:1:10: fatal error: Python.h: No such file or directory
1 | #include <Python.h>

mbedmain, 2022-06-20:
@sw-dev-code thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-06-21:
Hi @sw-dev-code 

can you create virtual env, install python 3.9 and install requirements and try compiling again? This looks like env issue.

It would also help to provide versions of installed packages (use pip freeze or similar to print all versions of requirements).

sw-dev-code, 2022-06-21:
@0xc0170 I will try with 3.9. Also, with python3.8 I did it in venv and got the same result. Please check the pip freeze output.

```
(venv) asus@ASUS:~/git/node$ python -m pip freeze
appdirs==1.4.4
asn1ate==0.6.0
beautifulsoup4==4.6.3
cbor==1.0.0
certifi==2022.6.15
cffi==1.15.0
chardet==3.0.4
Click==7.0
cmsis-pack-manager==0.2.10
colorama==0.3.9
cryptography==2.9.2
ecdsa==0.17.0
fasteners==0.17.3
future==0.18.2
icetea==1.2.4
idna==2.7
intelhex==2.3.0
Jinja2==2.10.3
jsonmerge==1.8.0
jsonschema==2.6.0
junit-xml==1.8
lockfile==0.12.2
manifest-tool==1.5.2
MarkupSafe==2.0.1
mbed-cli==1.10.5
mbed-cloud-sdk==2.0.8
mbed-flasher==0.10.1
mbed-greentea==1.8.13
mbed-host-tests==1.8.13
mbed-ls==1.8.13
mbed-os-tools==1.8.13
milksnake==0.1.5
prettytable==2.5.0
protobuf==3.5.2.post1
psutil==5.6.6
pyasn1==0.2.3
pycparser==2.21
pycryptodome==3.14.1
pyelftools==0.25
pyparsing==3.0.9
pyserial==3.4
python-dateutil==2.8.2
python-dotenv==0.20.0
pyusb==1.2.1
PyYAML==4.2b1
requests==2.20.1
semver==2.13.0
six==1.12.0
urllib3==1.24.2
wcwidth==0.2.5
yattag==1.14.0
```
Also, you can see that Python.h exists.
```
ls -lh /usr/include/python3.8/ | grep Python.h
-rw-r--r-- 1 root root 3.6K Mar 15 13:22 Python.h
```

0xc0170, 2022-07-04:
I can try to reproduce the error, but will need to know the exact steps to use. We havent seen this error in CI.

What example are you compiling ? Have you tried any other application?

amrudesh-santhanam, 2022-11-12:
I have the same issue with python3.8.10 on Ubuntu 20.04 even with the blinky library I got by  `mbed import https://github.com/ARMmbed/mbed-os-example-blinky#mbed-os-5.15.9 my-blinky`