ARMmbed/mbed-os #15304: ME910G1 VS HE910DG

AwetCarpigiani, 2022-07-01:
Hello Guys!

I have mounted the ME910G1 chip and HE910DG chip in two identical  circiut boards while the ME910G1 works fine the HE910DG is not working.
What could it be? 

Thank you

mbedmain, 2022-07-01:
@AwetCarpigiani thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-07-01:
This is not the issue, neither related to Mbed OS. I'll close this.

closed 2022-07-01 by 0xc0170