ARMmbed/mbed-os #15307: Arduino Nano 33 BLE stops execution after 8.5 minutes when LPTICKER is enabled

drewmnoel, 2022-07-05:
### Description of defect

Arduino Nano 33 BLE (`nRF52840` based) stops execution without error on stdout after about 8.5 minutes unless `LPTICKER` is disabled. With `LPTICKER` disabled (e.g. through `target.device_has_remove`), applications appear to run without issue, or at least for several hours.

Debugging with OpenOCD shows that after that time, breakpoints in the program do not execute. RTC interrupts continue to execute, and can be stepped through: https://github.com/ARMmbed/mbed-os/blob/91b793c4fd1ad7f02b8db5a2e0d77daf63e5c836/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/common_rtc.c#L81-L97

Hypothesis: since the `nRF52840` uses an RTC with a 24-bit counter and frequency of 32768 Hz, the RTC overflow is at around 512 seconds (about 8.5 minutes), it's a possible culprit.

#### Target(s) affected by this defect ?
`ARDUINO_NANO33BLE` (which is `nRF52840` based).

Bootloader information below from `bossac`:
```
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
```

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, version 12.1.0

#### What version of Mbed-os are you using (tag or sha) ?

https://github.com/ARMmbed/mbed-os/commit/91b793c4fd1ad7f02b8db5a2e0d77daf63e5c836

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
* mbed CLI 1.10.4, made a copy with `mbed-tools` v7.55.0 but the behavior is the same.
* [Arduino's `bossa`](https://github.com/arduino/bossa)

#### How is this defect reproduced ? 
Clone the [sample project](https://github.com/drewmnoel/nRF52840-lockup), remove the `device_has_remove` line from `mbed_app.json` and compile. Flash to board and execute. After approximately 8.5 minutes, the program appears to halt without any errors on the console line.

To remediate, re-add the `device_has_remove` line from `mbed_app.json` and do a fresh compile. Flash to board and execute. Application runs at least more than a few hours without issue.

0xc0170, 2022-07-06:
cc @facchinm

drewmnoel, 2022-07-06:
Sorry, I'm not sure what you mean. LPTICKER has been removed from something? It's used by default for my target unless I manually override that in my `mbed_app.json`, seen in the MCU_NRF52840 targets list: https://github.com/ARMmbed/mbed-os/blob/91b793c4fd1ad7f02b8db5a2e0d77daf63e5c836/targets/targets.json#L6865-L6889

I scanned a bunch of pull requests related to nrf52 and LPTICKER, and commits for TARGET_NRF5x going back 2 years but didn't see anything about LPTICKER support being removed or known-broken etc, can you point me to one in particular? If, ultimately, support _is_ broken, the `targets.json` should likely be updated to not use it by default.

0xc0170, 2022-07-06:
I did not fully followed your issue report. I got confused "when enabled".

I understand now, the example shared does not work with Mbed OS on Arduino Nano 33 target with no changes.

facchinm, 2022-07-07:
Hi @drewmnoel ,
the default bootloader "forgets" to deinitialize all the peripherals it uses before jumping to the actual sketch.
If you keep the bootloader, these lines https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/ARDUINO_NANO33BLE/variant.cpp#L139-L178 should be added to ensure the boards state is consistent.

drewmnoel, 2022-07-07:
@facchinm that _exactly_ fixes the issue and explains the behavior I'm seeing. I don't expect mbed to "know" to clear that interrupt, so it probably makes sense to close this issue then. Thanks for the help!

closed 2022-07-22 by 0xc0170