ARMmbed/mbed-os #15309: Wrong algorithm for CAN timing settings in can_api.c (STM32G474RE)

chdelfs, 2022-07-23:
### Description of defect

#### Target(s) affected by this defect ?
     I faced this issue on NucleoG474RE board, but likely all STM32G4 boards are affected.

#### Toolchain(s) (name and version) displaying this defect ?
     GCC 10.2.1 arm-none-eabi

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cmake

#### How is this defect reproduced ?

mbedmain, 2022-07-23:
@chdelfs thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What Mbed OS version are you using?
How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

chdelfs, 2022-07-23:
My suggestion is that can_api.c is modified in line 312:
Change from:
`  while (!IS_FDCAN_DATA_TSEG1(ntq / nominalPrescaler)) {`
to:
`  while (!IS_FDCAN_NOMINAL_TSEG1(ntq / nominalPrescaler)) {`

After changing the Mbed source as above, I could successfully use CAN bus on NucleoG474RE.

0xc0170, 2022-08-03:
cc @ARMmbed/team-st-mcd

0xc0170, 2022-08-03:
> My suggestion is that can_api.c is modified in line 312:

Would you send a pull request addressing this?

chdelfs, 2022-08-03:
I will do my very best..

jeromecoutant, 2022-08-16:
Hi
#15317 approved
Thx @chdelfs