ARMmbed/mbed-os #15318: FileSecurityDb::sync does not write data to file on FlashIAPBlockDevice

mkj-stonehill, 2022-08-04:
### Description of defect

I am using FlashIAPBlockDevice to hold a LittleFileSystem, and trying to use a file stored there to hold the BLE security database.  The file does indeed get created when SecurityManager::init() is called; however, no data is ever saved to the file, even after SecurityManager::writeBondingStateToPersistentStorage() is called.

I can see traces which indicate that _db->sync was called, and the valid, connected, entry was written to the file, and flush was called.  However, after a reset/power cycle, the file is still 0 bytes long (and so just gets re-initialized as empty).  I believe the, although FileSecurityDb::sync() does write out the entries, and then calls fflush(_db_file), that does not actually trigger the data to be written to flash.

On Linux, you would normally follow the fflush call with fsync(fileno(_db_file)); however, the "fileno" call does not exist in this environment.  The only way I can see to actually flush the file out to flash would be to close it, and then open it again.  This *could* be accomplished by calling SecurityManager::reset(), which closes the database; however, in my device, there's no logical place I can do that.

#### Target(s) affected by this defect ?

I am using Sparkfun Artemis Nano board, which has an Ambiq Apollo Blue MCU.  But I believe this would happen on any MCU using the FlashIAPBlockDevice to host a file system.

#### Toolchain(s) (name and version) displaying this defect ?

Mbed Studio 1.4.4
(This target uses the GCC_ARM toolchain)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99
Master branch SHA 9ca538338aef6a9b136aaca9b3dab000ceb6aa6e

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

I can reproduce the same effect by creating a file, writing to to, calling fflush, but never closing the file.  If I uncomment the fclose() call (and comment out the fflush() call), the file will contain the data written.

```
/** This will use blocks at the end of internal flash:
 *      64KB from 0xE_E000 to 0xF_E000
 *     128KB from 0xD_E000 to 0xF_E000
 */
BlockDevice *fsStorage = new FlashIAPBlockDevice(0x000DE000, 0x20000);

/** Filesystem for storing persistent information
 */
LittleFileSystem fs("fs");

main()
{
    int err = fs.mount(fsStorage);
    if (err)
    {
        // No valid filesystem present yet.  Format it.
        ble_tr_warn("Filesystem not found; formatting...");
        err = fs.reformat(fsStorage);
        if (err)
            error("Unable to format filesystem; error %d (%s)\n", err, strerror(-err));
    }

    const char data[] = "0123456789A";
    size_t dataLen = sizeof(data);
    FILE *f = fopen("/fs/test", "rb+");
    if (f != nullptr)
    {
        fseek(f, 0L, SEEK_END);
        tr_info("Test file is %u bytes", ftell(fp));
    }
    else
    {
        tr_error("Test file does not exist; try creating it");
        if (!(f = fopen("/fs/test", "wb+")))
            tr_error("Unable to create test file");
    }
    if (f)
    {
        int result = fwrite(data, 1, dataLen, f);
        if (result != dataLen)
            tr_error("Failed to write string; result=%d, expected=%d", result, dataLen);
        //fclose(f);
        fflush(f);
        
        tr_info("Successully wrote %d bytes to test file (just flushed; not closed)", dataLen);
    }
}

```

mbedmain, 2022-08-04:
@mkj-stonehill thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

mkj-stonehill, 2022-08-04:
If it were me, I would implement FileSecurityDb to keep the file name, instead of an open file handle.  When access to the file is required (i.e. when restoring the DB, or when updating the DB), I would open the file, do the I/O, then close the file.  I could take a crack at that in my fork; but I wanted to record the issue first (in case there is some tricksy way of getting the flash driver to actually write the file that I don't know about).