ARMmbed/mbed-os #15319: Cannot build with mbed-os-5.12.4

daltonv, 2022-08-08:
### Description of defect

Unable to build my project the depends on mbed-os-5.12.4 because there seems to be an issue with the dependencies
of mbed-os-5.12.4. Running `mbed deploy` on my project gives me the following error:
```bash:
[mbed] WARNING: Program "asic-programmer-mbed" in "/home/runner/work/OC_CCE7102_CCE7103_ASIC_programmer/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed" does not use source control management.
       To fix this you should use "mbed new ." in the root of your program.
---
[mbed] ERROR: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pyyaml, jsonschema, mbed_cloud_sdk, jinja2, mbed_ls, mbed_host_tests, mbed_greentea, pyelftools, manifest_tool, icetea, pycryptodome, pyusb, cmsis_pack_manager
       You can install all missing modules by running "pip install -r requirements.txt" in "mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
[mbed] Working path "/home/runner/work/OC_CCE7102_CCE7103_ASIC_programmer/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed" (library)
[mbed] Program path "/home/runner/work/OC_CCE7102_CCE7103_ASIC_programmer/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed"
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #73f096399b4c
[mbed] Auto-installing missing Python modules (pyyaml, jsonschema, mbed_cloud_sdk, jinja2, mbed_ls, mbed_host_tests, mbed_greentea, pyelftools, manifest_tool, icetea, pycryptodome, pyusb, cmsis_pack_manager)...
``` 

I then tried running `pip install -r mbed-os/requirements.txt` manually and tracked the issue down to a dependency `manifest-tool==1.4.8` in the requirements file. I get the following error when trying to install: 
 
```bash
pip install manifest-tool==1.4.8
Collecting manifest-tool==1.4.8
  Using cached manifest-tool-1.4.8.tar.gz (64 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      Traceback (most recent call last):
        File "/tmp/pip-install-ibfnlknk/manifest-tool_655b3f52ef464165b8c1a84424d9db5f/setup.py", line 28, in <module>
          from pip._internal.download import PipSession
      ModuleNotFoundError: No module named 'pip._internal.download'

      During handling of the above exception, another exception occurred:

      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-ibfnlknk/manifest-tool_655b3f52ef464165b8c1a84424d9db5f/setup.py", line 30, in <module>
          from pip.req import parse_requirements
      ModuleNotFoundError: No module named 'pip.req'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.12.4

mbed-cli = 1.10.5
python = 3.10.5

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
* Create a python virtual environment and activate it  
* Clone mbed-os
* cd into mbed-os
* Call `mbed update mbed-os-5.12.4`
* Call `pip install -r requirements.txt`

mbedmain, 2022-08-08:
@daltonv thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 
NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

daltonv, 2022-08-08:
I've figured out something of a solution

I manually bumped the `manifest-tool` version to `1.5.0` in the requirements.txt. This allowed me to successfully run `pip install -r requirements.txt`

However I ran into a new problem when running `mbed complile`

```bash
mbed compile -t GCC_ARM -m NUCLEO_L4R5ZI
[mbed] Working path "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed" (library)
[mbed] Program path "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed"
Traceback (most recent call last):
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/make.py", line 37, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/targets/__init__.py", line 32, in <module>
    from tools.psa import find_secure_image
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/psa/__init__.py", line 23, in <module>
    from .generate_partition_code import manifests_discovery, generate_spm_code
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/psa/generate_partition_code.py", line 28, in <module>
    from tools.psa.mbed_spm_tfm_common import validate_partition_manifests, \
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/psa/mbed_spm_tfm_common.py", line 24, in <module>
    from jinja2 import Environment, FileSystemLoader, StrictUndefined
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/lib/python3.8/site-packages/jinja2/__init__.py", line 33, in <module>
    from jinja2.environment import Environment, Template
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/lib/python3.8/site-packages/jinja2/environment.py", line 15, in <module>
    from jinja2 import nodes
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/lib/python3.8/site-packages/jinja2/nodes.py", line 19, in <module>
    from jinja2.utils import Markup
  File "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/lib/python3.8/site-packages/jinja2/utils.py", line 642, in <module>
    from markupsafe import Markup, escape, soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/lib/python3.8/site-packages/markupsafe/__init__.py)
[mbed] ERROR: "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/bin/python3" returned error.
       Code: 1
       Path: "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed"
       Command: "/mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/.venv/bin/python3 -u /mnt/c/projects/repositories/OC_CCE7102_CCE7103_ASIC_programmer/asic-programmer-mbed/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L4R5ZI --source . --build ./BUILD/NUCLEO_L4R5ZI/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

Looks like this `MarkupSafe` library is some dependency of jinja2? Anyway I was able to fix this issue by forcing `MarkupSafe` to use `v2.0.1` instead of the auto installed `v2.1.1`. 

Now my project can build, but I need my own separate requirements.txt file ([requirements.txt](https://github.com/ARMmbed/mbed-os/files/9282930/requirements.txt)). If someone can reproduce both my original problem and my solution can we get an updated requirements.txt as a bug fix to `mbed-os-5.12.4`?

0xc0170, 2022-08-09:
Hi,

we do not support anymore mbed-os-5.12 branch.

The problem could be also changes in pip (if you downgrade to version 9 or so), manifest error should disappear.

Some users might find this helpful if they have the similar problem. I'll close this as won't fix.

closed 2022-08-09 by 0xc0170