ARMmbed/mbed-os #15327: Is mbed-os dead?

DBS06, 2022-08-26:
# Is mbed-os dead?
I know it is a little bit provocative formulated, but I am working on a project since 3-.4 years now which uses mbed-os and it seems like that mbed-os is not really maintained and supported anymore.

## What leads me to this assumption:

We are using the NRF52840 from Nordic and Cordio-Stack-Team and mbed-os team still not solved this really urgent and major issue https://github.com/ARMmbed/mbed-os/issues/10669. The problem with this issue is, that Cordio-Stack removes the LE from BLE. We fixed it as far as we were able to do it, but without access to Cordio we cant reduce the power consumption any further and actually we don't want to touch it.
Recently on the Embedded-World fair, one of my colleagues asked ARM if they are still supporting mbed-os and they said no the will not support and maintain it, same goes for Nordic. As everybody knows who are using a Nordic-Chip Nordic stopped the support years ago.

## Conclusion/Impact
We are more or less finished our project were we developed a platform based on mbed-os for the current and future products and the release will be next month. Nevertheless we would like to have a base platform which will be maintained, supported and which has a future. Therefore I am asking if mbed-os is dead?
If it is the case, we have to move to i.e. Zephyr or similar, which will be painful but sadly I think we have no other choice...

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2022-08-26:
@DBS06 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

DBS06, 2022-09-01:
No answer is also an answer... 
For me personally I am a little bit sad about it, that mbed-os seems dead. Actually mbed-os wasn't that bad, actually it was on a good way, but the BLE-Issue kills it. At least we were able to fix the BLE-Power-Consumption-Issue in our project, with some hacks.
Nevertheless I am still hoping for a reply.

rvasquez6089, 2022-09-01:
I don't believe it is dead. There was recently a release in Jun 20. I believe developer resources have been focused on other aspects of the mbed ecosystem. Online compiler and IOT features. I would be quite sad if mbed os was dead though. I use it professionally in MANY projects. It provides and useful and quick abstraction for many STM32 MCU's. 
@jeromecoutant Do you know of the future plans for mbed-os?

DBS06, 2022-09-14:
@rvasquez6089 I completely agree with you, but I think mbed-os seems dead and it is more or less dead, therefore I would recommend everyone who has the chance to switch to zephyr should do the switch.

yuyangworld, 2022-09-27:
是死是活倒是给个痛快话啊？

kalacool, 2024-07-15:
> 是死是活倒是给个痛快话啊？

It looks like it’s finally dead...
https://www.hackster.io/news/end-of-the-line-for-arm-s-mbed-os-7027e92805e7