ARMmbed/mbed-os #15331: STM32WLE5JC serial stop working when deepsleep

shiwzhi, 2022-09-15:
### Description of defect

#### Target(s) affected by this defect ?
STM32WLE5JC

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

807fd7965164737a1c08e1b50e710b2f854ef193

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.4.4.1 on Mac OS

#### How is this defect reproduced ? 
```
#include "mbed.h"

#define TARGET_TX_PIN                                                     PA_2
#define TARGET_RX_PIN                                                     PA_3

UnbufferedSerial serial_port(TARGET_TX_PIN, TARGET_RX_PIN, 9600);

int main(void)
{
    
    while(1) {
        serial_port.write("hello\n", 6);
        ThisThread::sleep_for(5s);     //  Serial stop working no output after
    }
    
}
```

```
#include "mbed.h"

#define TARGET_TX_PIN                                                     PA_2
#define TARGET_RX_PIN                                                     PA_3

int main(void)
{
    
    while(1) {
        UnbufferedSerial serial_port(TARGET_TX_PIN, TARGET_RX_PIN, 9600);   //reinit serial seems working
        serial_port.write("hello\n", 6);
        ThisThread::sleep_for(5s); 
    }
    
}
```

mbedmain, 2022-09-15:
@shiwzhi thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

stefan-npsys, 2022-09-15:
Just for info (could be related). I see the same on the SEED E5 (also STM32WLE5JC, GCC, MBED 6.14) on the console stdio.

The workaround for me is to disable IO before sleep and enable after sleep. Something like:

```
mbed_file_handle(STDIN_FILENO)->enable_input(false);
mbed_file_handle(STDOUT_FILENO)->enable_output(false);
  ThisThread::sleep_for(5s);
mbed_file_handle(STDIN_FILENO)->enable_input(true);
mbed_file_handle(STDOUT_FILENO)->enable_output(true);
```

hallard, 2022-09-20:
Guys, this may explains why console is working on boot then all `printf` in the loRaWAN stuff does sort nothing.
But in my case as far as I remember looks like it is happening only in release mode. Need to check back this one 
Would be great to have this one solved, @jeromecoutant any idea?

0xc0170, 2022-09-26:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2022-11-14:
> MBED pulls the MCU into STOP-mode 2 on deep-sleep (and not STOP-1 as expected with most things retained)

Yes, Stop2 is used:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/sleep.c#L200

hallard, 2023-01-02:
Any idea on how to fix this one, for information same code with Murata target (STM32L0) does not have this issue, but on STM32WL wake up after sleep does not works

As simple as 
```cpp
int main()
{
    #if defined (MBED_DEBUG)
    const uint8_t device_mode = 2 ; // Debug
    #elif defined (NDEBUG)
    const uint8_t device_mode = 0 ; // Release
    #else
    const uint8_t device_mode = 1 ; // Develop
    #endif
    int n = 100;
    printf(CRLF CRLF);
    printf("Hello World!"CRLF);
    printf(CRLF);
    printf("Compilation mode %s" CRLF, device_mode==2 ? "DEBUG" : device_mode==0 ? "RELEASE" : "DEVELOP");
    while (n--) {
        printf("Loop %d" CRLF, n);
        ThisThread::sleep_for(500ms);
    }
}
```

**DEBUG**
```
Hello World!

Compilation mode DEBUG
Loop 99
Loop 98
Loop 97
Loop 96
Loop 95
Loop 94
Loop 93
Loop 92
```

All other stuck with Serial not working anymore after sleep

**RELEASE**
```
Hello World!

Compilation mode RELEASE
Loop 99
```

**DEVELOP**
```
Hello World!

Compilation mode DEVELOP
Loop 99
```

shiwzhi, 2023-01-06:
Issue solved after set baudrate to 9600 and use lpuart pin
PA_2_ALT0、PA_3_ALT0 instead
```
"GENERIC_STM32WLE5JC": {
            "stm32wl-lora-driver.crystal_select": 0,
            "target.stdio_uart_tx": "PA_2_ALT0",
            "target.stdio_uart_rx": "PA_3_ALT0"
 }

closed 2023-01-06 by shiwzhi

JojoS62, 2023-01-06:
a difference is that debug build does not enter deep sleep mode. You can try also to lock deep sleep:
https://os.mbed.com/docs/mbed-os/v6.15/apis/power-apis.html

hallard, 2023-01-06:
> Issue solved after set baudrate to 9600 and use lpuart pin PA_2_ALT0、PA_3_ALT0 instead
> 
> ```
> "GENERIC_STM32WLE5JC": {
>             "stm32wl-lora-driver.crystal_select": 0,
>             "target.stdio_uart_tx": "PA_2_ALT0",
>             "target.stdio_uart_rx": "PA_3_ALT0"
>  }
> ```

Indeed does works (on this sample I need to check with my LoRaWAN test), amazing but from my point of view it's just a workaround not a real fix, and tried with 19200 does not works either and this is not looking a "normal thing"

any idea from mbed guru's @jeromecoutant @0xc0170 ?

jeromecoutant, 2023-01-09:
Hi
Maybe check table 37 from STM32WLEx Reference Manual:
Mbed-os deepsleep is ST stop 2 mode, and there is some difference between USART1&2 and LPUART

hallard, 2023-01-09:
Yeah makes sense difference between LP_UART and USART (like all ST) and I have no problem with this. What I don't understand (and I can't find any referenece) is why when we wake from deep sleep (I except wake from LP_UART of course) from any GPIO or RTC for example, why console works back at 9600 and not 115200, does this means that STOP2 does not retain baud rate and waking up reset LP_UART default speed to 9600?

hallard, 2023-02-23:
Sorry for revival but looking at documentation LPUART1 is able to wake from STOP2, 

[image]

I'm using LPUART1 and it works at 9600 but only at 9600 any hint to search place why at other speed wake from sleep stop console working and I see nothing in datasheet indicating that it should not work?

jeromecoutant, 2023-02-24:
Hi
About LPUART and low power mode, I know:
- there is some limitation depending on clock source: see 6.2.21 Peripheral clocks enable => HSI16, LSI, or LSE
- there is limitation for baud rate: see 34.1 LPUART introduction 
=> Only 32.768kHz LSE clock is required to enable UART communications up to 9600 baud/s. Higher baud rates can be reached when the LPUART is clocked by clock sources different from the LSE clock

hallard, 2023-02-24:
Thanks for information, interesting.
Do you know where in mbed-os is located stuff done on wake-up for serial console?
I think we may need to add something for this target to reconfigure back console (in case of using LPUART) to baud rate (or other parameter) it was before sleeping. I can take a look and do some test if I have any entry point.

Thanks

hallard, 2023-02-24:
FYI just tried these 3 options without any changes

```json
"target.lpuart_clock_source": "USE_LPUART_CLK_HSI",
"target.lpuart_clock_source": "USE_LPUART_CLK_LSE",
"target.lpuart_clock_source": "USE_LPUART_CLK_PCLK1",
```

hallard, 2023-05-02:
Just found this document, just putting here as a reminder, interesting 

[ultralowpower of stm32wl](https://www.st.com/resource/en/application_note/an5568-ultralowpower-features-of-stm32wl-series-microcontrollers-stmicroelectronics.pdf)

m0rphex, 2023-08-25:
With the e5 mini I'm trying transmit at 1200 baud after deep sleep without success. 4800 baud works. Any insight on how to get the slower baud working would be much appreciated.