ARMmbed/mbed-os #15334: TARGET Maxim get_usb_phy() issue due to incorrect target specified in USBPhy_Maxim.cpp

Gerriko, 2022-10-06:
### Description of defect
I attempted to run this USB Serial example (https://os.mbed.com/teams/MaximIntegrated/code/FTHR_USB_Serial_mbed-v67/) in Mbed Studio using Mbed OS 6.16 (and Mbed OS 6.15) with a MAX32620FTHR board.

The example compiles but creates a run error as it cannot find a hardware USB driver.
This is resolved by adding USBDEVICE as a target override in mbed_app.json.

However, this now creates a known compile problem, namely `Undefined symbol get_usb_phy()`

Upon further online searches, I checked the TARGET Maxim folder and I could see that both USBPhyHw.h and USBPhy_Maxim.cpp files were present as they should be. So this did not make sense.

I opened up USBPhy_Maxim.cpp file and spotted a problem with the following compiler directive as there is no TARGET_MAX32620 in the folder: 

```
#if defined(DEVICE_USBDEVICE) && (DEVICE_USBDEVICE) && \
    (defined(TARGET_MAX32620) || defined(TARGET_MAX32625) || defined(TARGET_MAX32630))

```
The folder is listed as TARGET_MAX32620C

Thus by changing all cases with TARGET_MAX32620 listed to TARGET_MAX32620C in that file, it was able to compile.

I am reporting an error in the file USBPhy_Maxim.cpp, which incorrectly uses **TARGET_MAX32620** instead of **TARGET_MAX32620C**

#### Target(s) affected by this defect ?

All Maxim MAX32620xxxx boards

#### Toolchain(s) (name and version) displaying this defect ?

ARM Compiler | 6.16 | up to date |  

 

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 6.15.1.
Mbed OS 6.16.0.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio: 1.4.4
Clangd | 12.0.2 | up to date |  
Example projects | 1.11.0 | up to date |  
GDB client | 6-2017-q2-update | up to date |  
Git | 2.23.1 | up to date |  
Mbed Library Cache | 1.10.0 | up to date |  
Python tools | 0.6.0 | up to date |  
Debug Packs | 1.1.0 | up to date

#### How is this defect reproduced ? 
I attempted to run this USB Serial example (https://os.mbed.com/teams/MaximIntegrated/code/FTHR_USB_Serial_mbed-v67/) in Mbed Studio using Mbed OS 6.16 (and Mbed OS 6.15) with a MAX32620FTHR board.

0xc0170, 2022-10-10:
cc @ARMmbed/team-maximintegrated please review

ahmetalincak, 2022-10-12:
Hi @Gerriko,

Thank you for raising the issue. We'll investigate and apply the solution ASAP. You can also create a Pull Request for your solution.

Regards,

Ahmet.