ARMmbed/mbed-os #15335: __ARM_ARCH Macros are not defined

BioBox, 2022-10-13:
### Description of defect

Obviously the operating system needs to know the version of ARM that we're using, but the compiler doesn't know because the `ARM_ARCH_*` macro is never set even though there's one set for the specific processor. I think the easiest way to fix this is to add it into the `.cmake` files found in `tools/cmake/cores/`

#### Target(s) affected by this defect ?

All of them.

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.16.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli2

#### How is this defect reproduced ? 

I'm not the only one experiencing this. #15216 

`mbedtools compile --mbed-os-path ../../os/mbed-os -t GCC_ARM -m LPC1768`

0xc0170, 2022-10-14:
cmsis files take care of defining architecture macros if they are not already defined by a compiler, see https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/cmsis_iccarm.h#L59

why we should define them?

BioBox, 2022-10-14:
Uh, because it doesn't work without it?

0xc0170, 2022-10-18:
> Uh, because it doesn't work without it?

You can describe how to reproduce the issue. Mbed Os requires cmsis headers so a target should get the files required to get these architecture macros. It's par of CMSIS 5 project, they are maintained there.

We do not want to duplicate what is already provided.

closed 2022-10-21 by BioBox