ARMmbed/mbed-os #15338: crash log show _free_r on mimxrt1060

CamelShoko, 2022-10-26:
### Description of defect

On platform MIMXRT1062.

have some HardFault error. use the crash_log_parser.py checked all shows the _free_r

```
Crash Info:
	Crash location = _free_r [0x600C6B84] (based on PC value)
	Caller location = osMutexAcquire [0x6005BC73] (based on LR value)
	Stack Pointer at the time of crash = [202504D0]
	Target and Fault Info:
		Processor Arch: ARM-V7M or above
		Processor Variant: C27
		Forced exception, a fault with configurable priority has been escalated to HardFault
		A precise data access error has occurred. Faulting address: 404646F

```

#### Target(s) affected by this defect ?
MIMXRT1062

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
version 6.16.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
some time.

CamelShoko, 2022-10-26:
more:

MBED_HEAP_STATS_ENABLED = 1

0xc0170, 2022-10-26:
@CamelShoko you should provide more details how to reproduce a problem. It would be best to identify how to crash the logger.

cc @ARMmbed/team-nxp

CamelShoko, 2022-11-04:
it maybe a NXP cache issues.
i do a cache clean in code, the problem solved.