ARMmbed/mbed-os #15348: EMAC driver initializes an incorrect pin (PG2) on the STM32F4 -> NUCLEO_F429ZI target

akiroz, 2022-11-17:
### Description of defect

EMAC driver initializes an incorrect pin (PG2) on the STM32F4 -> NUCLEO_F429ZI target

In mbed 5.15, the EMAC driver initializes PG2 for RMII RX ER on STM32F429, this pin is not part of the RMII interface for this chip.

See: https://github.com/ARMmbed/mbed-os/blob/bacf37894945a5c9b7619b709eeabeeb68d25d9c/features/netsocket/emac-drivers/TARGET_STM/TARGET_STM32F4/TARGET_NUCLEO_F429ZI/stm32f4_eth_init.c#L54

The correct pin for the RMII RX ER signal is on PB10, however this signal is not connected on the Nucleo development board.

#### Target(s) affected by this defect ?

STM32F4 -> NUCLEO_F429ZI

#### Toolchain(s) (name and version) displaying this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

N/A

jeromecoutant, 2022-11-17:
Hi
I checked User Manual for this Nucleo, and there is no RMII RX ER (not on PG2, not in PB10)...

akiroz, 2022-11-17:
Yes, the RMII RX ER signal is not routed in the Nucleo board. PB10 is where the MCU would support this signal if used.

akiroz, 2022-11-17:
Since this signal is optional and not routed in Nucleo, I think an easy fix is simply deleting PG2 from the init code.
I checked a few other STM targets and they also seem to be affected by this issue.

jeromecoutant, 2022-11-17:
Thx, please propose a patch in order to review and correct :-)

PR #15358 (2023-02-01): fix(drivers/emac): [mbed6] Remove incorrect RMII RX ER initialization
### Summary of changes 

Fixes #15348 
master based PR of #15349

#### Affected targets:

- ST Nucleo MB1137: Signal not routed (Ref: DocID028599 Rev 7)
  - TARGET_NUCLEO_F207ZG
  - TARGET_NUCLEO_F429ZI
  - TARGET_NUCLEO_F439ZI
  - TARGET_NUCLEO_F746ZG
  - TARGET_NUCLEO_F756ZG
  - TARGET_NUCLEO_F767ZI
- ST 32F746GDISCOVERY: RX_ER signal is on PB10/PI10 not PG2 (Ref: DocID027590 Rev 4)
  - TARGET_DISCO_F746NG
- ST 32F746GDISCOVERY: RX_ER signal is on PB10/PI10 not PD5 (Ref: DS11532 Rev 7)
  - TARGET_DISCO_F769NI

#### Impact  [... cut]
closed 2023-02-01 by saheerb