ARMmbed/mbed-os #15351: keil studio: m2354: build failed

ccli8, 2022-11-23:
### Description of defect

On Keil Studio, on target NU_M2354, mbed-os-example-blinky builds failed.
On Mbed Studio or Mbed CLI, build OK

#### Target(s) affected by this defect ?
NU_M2354

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.16.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Keil Studio

#### How is this defect reproduced ? 
1. Log in Keil Studio
1. Click File > Clone and specify URL https://github.com/ARMmbed/mbed-os-example-blinky
1. Switch to mbed-os-6.16.0 if not yet
1. Specify NUMAKER-M2354 as target hardware
1. Build failed with log:
```
Build started
Using toolchain ARMC6 profile {'ENV': {'ARMLMD_LICENSE_FILE': '8224@10.100.118.115:8224@10.100.172.27'}, 'PATHS': {'ARMC6_PATH': '/opt/ARMCompiler6.15.13/bin/', 'ARM_PATH': '/opt/armcc5_06_u6/'}, 'common': ['-c', '--gnu', '-O3', '-Otime', '--split_sections', '--apcs=interwork'], 'cxx': ['--cpp', '--no_rtti'], 'COMPILE_C_AS_CPP': False, 'NEW_SCAN_RESOURCES': True}
scan /tmp/chroots/ch-9559ea1a-25c0-4b7f-8ee0-9d5ca920a19b/src
scan /tmp/chroots/ch-9559ea1a-25c0-4b7f-8ee0-9d5ca920a19b/extras/mbed-os.lib
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
......
elf2bin mbed-os-example-blinky.NU_M2354
Target not supported: Missing post-build hook
Internal error.
Build failed
```

mbedmain, 2022-11-23:
@ccli8 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2022-11-24:
@arekzaluski would you review please?

arekzaluski, 2022-12-05:
@ccli8 thank you for reporting this issue. I can reproduce it in both Keil Studio Cloud and Online Compiler. Let us investigate why it happens.

lunchboxwu, 2023-05-01:
Is this issue solved? I encountered the same issue on Nuvoton M2354 on mbed-os-6.17.0

ccli8, 2023-05-02:
@arekzaluski I just check the issue and it remains there. Would you help check it?