ARMmbed/mbed-os #15372: wi-sun stack ws_bootstrap_neighbor_info_get return one more info

mcuxmx, 2023-01-13:
### Description of defect

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.16.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
When the input parameter table_max of the function ws_bootstrap_neighbor_info_get is smaller than the number of neighbor tables (mac_neighbor_info(cur)->list_total_size), the function will return one more neighbor information.
For example: when table_max=16, the current number of neighbors is 20 mac_neighbor_info(cur)->list_total_size=20, the function will return 17 neighbor information.

https://github.com/ARMmbed/mbed-os/blob/2e506ba7935519e04917557aab5b3176e41bbd5f/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_bootstrap.c#L2540-L2543
The expression “count > table_max” probably should be “count >= table_max”

mbedmain, 2023-01-13:
@mcuxmx thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

mcuxmx, 2023-01-13:
also, wouldn't it be better to define the data type of variable "count" in line 2524 as uint16_t?

https://github.com/ARMmbed/mbed-os/blob/2e506ba7935519e04917557aab5b3176e41bbd5f/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_bootstrap.c#L2522-L2526