ARMmbed/mbed-os #15377: target STM32F303CC has wrong RAM size

JojoS62, 2023-01-22:
### Description of defect

https://github.com/ARMmbed/mbed-os/blob/1dd0bce88c39bff1c86a3c2382087d0a6a40c766/targets/TARGET_STM/TARGET_STM32F3/TARGET_STM32F303xC/cmsis_nvic.h#L33

this MCU has 40 kB RAM in this region, not 48 kB. This leads to setting the stackpointer into nirvana and the target fails to start and even crashes gdb.
The setting is als wrong in the ARM pack manager IRAM1 section.

#### Target(s) affected by this defect ?

targets that depend on this MCU, STM32F303CC

#### Toolchain(s) (name and version) displaying this defect ?

all, RAM_SIZE is used in gcc, arm, iar

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.16.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2023-01-22:
@JojoS62 thank you for raising this issue.Please take a look at the following comments:

It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2023-01-23:
Hi

> The setting is als wrong in the ARM pack manager IRAM1 section.

I confirm: wrong info in keil pack => wrong info in ARM pack manager IRAM1 mbed section => wrong info in cmsis_nvic.h file

I  am correcting this now

PR #15378 (2023-01-26): STM32F303xC: correct RAM size
### Summary of changes 

Fix #15377 

wrong info in keil pack 
  => wrong info in ARM pack manager IRAM1 mbed section
  => wrong info in cmsis_nvic.h file

This PR corrects the mbed-os part :-)

@JojoS62

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2023-01-26 by 0xc0170