ARMmbed/mbed-os #15379: I2C STM32WL Debug : Unable to link because inline are not inlined

hallard, 2023-01-28:
### Description of defect

Depending on code, sometimes unable to link using I2C on STM32WL because looks like inline not inlined

#### Target(s) affected by this defect ?

STM32WL (may be other)

#### Toolchain(s) (name and version) displaying this defect ?

[image]

#### What version of Mbed-os are you using (tag or sha) ?

 6.16.0 and master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed studio 1.4.4.1

#### How is this defect reproduced ? 

Not really sure sometimes in happens with some code other not. 
I suspect compiler defect under certain circonstances depending on code, in my case target use `mbed-cryptoauthlib` for I2C secure element and another I2C device. in that case it always happen (in `debug` only the one that do not have `--inline` flag on profile)
`develop` and `release`profiles  have no issues

In my case `i2c_is_ready_for_transaction_start()` is not created (not inlined) failed to link (makes sense)

```
[Error] @0,0: L6218E: Undefined symbol i2c_is_ready_for_transaction_start (referred from BUILD/EVOLY_T_V12/ARMC6/mbed-os/targets/TARGET_STM/i2c_api.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol i2c_is_ready_for_transaction_start (referred from BUILD/EVOLY_T_V12/ARMC6/mbed-os/targets/TARGET_STM/i2c_api.o).
Finished: 0 information, 1 warning and 1 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol i2c_is_ready_for_transaction_start (referred from BUILD/EVOLY_T_V12/ARMC6/mbed-os/targets/TARGET_STM/i2c_api.o).
Finished: 0 information, 1 warning and 1 error messages.
```

anyway changing to always_inline with solve the issue (not sure it's the best option)

```cpp
inline bool i2c_is_ready_for_transaction_start(stm_i2c_state state)
{
    // Note: We can safely send a transaction start in the middle of any single byte operation; this creates a
    // repeated start.

    return state == STM_I2C_IDLE || STM_I2C_PENDING_START == state
           || state == STM_I2C_SB_READ_IN_PROGRESS || state == STM_I2C_SB_WRITE_IN_PROGRESS;
}
```

```cpp
MBED_FORCEINLINE bool i2c_is_ready_for_transaction_start(stm_i2c_state state)
```

mbedmain, 2023-01-28:
@hallard thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

multiplemonomials, 2023-02-03:
Wow that's odd, it sorta feels like an Arm Compiler internal issue

0xc0170, 2023-02-06:
I've seen this before somewhere. I found this page now that could help: 

https://developer.arm.com/documentation/ka002210/latest

PR #15383 (2023-02-23): fix STM I2C compiler inline issue

### Summary of changes 

fixed #15379

### Documentation 

removed inline that trigger link issues on STM32WL as @0xc0170 proposal in #15379

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Return code change / API behaviour change)

--------------------------------------------- [... cut]
closed 2023-02-23 by 0xc0170