ARMmbed/mbed-os #15380: LPC546XX / LPC54628 missing CAN feature?

mxcop, 2023-01-31:
### Description of defect
The LPC546XX Target includes the LPC54628 board which should have CAN as one of it's features.
At the moment I'm unable to use the mbed CAN api because I believe it thinks it doesn't have this feature.
*(I'm not sure if there are more boards that fall under LPC546XX that also can use CAN)*

#### Target(s) affected by this defect ?
LPC546XX / LPC54628

#### Toolchain(s) (name and version) displaying this defect ?
`toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)`

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.9.0
`framework-mbed @ 6.60900.220911 (6.9.0)`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
platform-io-6.1.6

#### How is this defect reproduced ? 
Creating a new platformio project with this config :
```
platform = nxplpc
board = lpc546xx
framework = mbed
```
If you try to use the CAN api it will just error saying it is undefined.

0xc0170, 2023-02-02:
There's can driver coming from their SDK in the codebase but it was not ported.

You can port it and create a pull request.

mxcop, 2023-03-06:
Where in the repo should the drivers be located?
I'm having some difficulty finding where to put my implementation.

I've mostly figured out how to use their CAN driver, so I might implement it for mbed-os.
*(If I find the time for it :p)*

0xc0170, 2023-06-13:
I am closing this as the question was answered. THe driver was not ported.

closed 2023-06-13 by 0xc0170