ARMmbed/mbed-os #15384: ZLP not sent in USB device

daniel-starke, 2023-02-24:
### Description of defect

Sending out a 64 byte buffer on USB CDC does not show up on the receiving side until more data with a length != 64 bytes is sent.
The USB protocol is a physical 1:1 half-duplex bus protocol with multiple channels (a.k.a. endpoints). That means only one side can send at a time. The device defines a buffer size for each endpoint. Any data larger than this needs to be split into multiple packets. To form a transaction and block the bus until end of the data transmission pakets with the maximum size for the endpoint are sent. The end of the transaction is signalled by sending a packet with less than the maximum size. A zero-length packet (ZLP) can be sent in case no more data is available but the complete data size was a multiple of the maximum endpoint buffer size.
STM32 does not automatically sends out ZLPs at the end of a trancaction as it has no knowledge about it. This needs to be done in the USB device driver. The user can not do so as the driver prevents transmissions of zero length. See line 402

https://github.com/ARMmbed/mbed-os/blob/17dc3dc2e6e2817a8bd3df62f38583319f0e4fed/drivers/usb/source/USBCDC.cpp#L398-L407

As this appears to be a USB protocol issue instead of a STM32 HAL specific one I would suggest solving it in

https://github.com/ARMmbed/mbed-os/blob/17dc3dc2e6e2817a8bd3df62f38583319f0e4fed/drivers/usb/source/USBDevice.cpp#L990-L1006

However, I am not aware of how other device platforms handle this. It may be necessary to send a ZLP if the last packet was a multiple of the [maximum endpoint buffer size](https://github.com/ARMmbed/mbed-os/blob/17dc3dc2e6e2817a8bd3df62f38583319f0e4fed/targets/TARGET_STM/USBPhy_STM32.cpp#L50) and track its transmission state in

https://github.com/ARMmbed/mbed-os/blob/17dc3dc2e6e2817a8bd3df62f38583319f0e4fed/targets/TARGET_STM/USBPhy_STM32.cpp#L108-L119

#### Target(s) affected by this defect ?

STM32

#### Toolchain(s) (name and version) displaying this defect ?

PlatformIO latest version

#### What version of Mbed-os are you using (tag or sha) ?

 mbed-os-6.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

PlatformIO latest version

#### How is this defect reproduced ? 

Send out 64 bytes after a client connects to a USB CDC port. Read from it on the client side (e.g. by opening the serial device with PuTTY). The 64 bytes of data should show up but don't.

0xc0170, 2023-03-02:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2023-03-02:
Hi @daniel-starke 
I think if you could propose a patch in USB drivers source and/or in STM port, it will help  lot!
Thx

daniel-starke, 2023-03-02:
The easiest solution is to disallow any write with a size equal to the endpoint buffer size. However, this solution may not be desired.
I could create a correct patch for the STM port but only if it is clear that it needs to be fixed there and not in the generic driver part.

cyliangtw, 2023-03-08:
@daniel-starke , some USB device may have the capability to detcect packet size & auto send ZLP. 
So, it's better to create the patch for STM port only instead of generic part.

agausmann, 2023-09-06:
I also ran into this, and put together a reproduction (#15451)

jeromecoutant, 2023-11-20:
Hi

@cyliangtw @agausmann Could you have a look on #15465 ?

Thx

cyliangtw, 2023-11-24:
> Hi
> 
> @cyliangtw @agausmann Could you have a look on #15465 ?
> 
> Thx

About usbd control packet, https://github.com/ARMmbed/mbed-os/blob/72f27cee92a4bdafa30513f732e007ba635dc93f/drivers/usb/source/USBDevice.cpp#L262C4-L262C4 could handle & send ZLP.

About usbd non-control packet, it's possible to wait ZLP.
However let H/W driver auto send ZLP, it's possible double ZLP in control packet case and may impact the next USBH IN token.
So, I revised my opinion, let USBCDC to determine whether or not to send ZLP should be the better choice.

daniel-starke, 2023-12-25:
Pull request #15473 has been merged. Therefore, this issue can be closed.

closed 2023-12-28 by 0xc0170