ARMmbed/mbed-os #15386: Add fully buildable support for STM32H745

jtmyz9, 2023-03-08:
### Description of defect

From https://github.com/ARMmbed/mbed-os/pull/12801 , base support for the STM32H745 was added, but is not a buildable target as is. Remaining configuration/PinNames/PeripherialPins files need to be add to make this a buildable target. 

#### Target(s) affected by this defect ?
STM32H745 / NUCLEO-H745ZI

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
6.17.0

mbedmain, 2023-03-08:
@jtmyz9 thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

cyliangtw, 2023-03-08:
@jtmyz9 , If board maker not implement the specific target, you could refer to [custom target porting guide](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/custom-target-porting.md) based on inherits MCU_STM32H745xI_CM4 or MCU_STM32H745xI_CM7.

For board level, to try adding custom_targets.json & PinNames.h, then build it.

For ex:  use case NU_IOT_M2354 [custom_targets.json](https://github.com/OpenNuvoton/NuMaker-mbed-OS-6-AWS-IoT/blob/master/custom_targets.json); [PinNames.h](https://github.com/OpenNuvoton/NuMaker-mbed-OS-6-AWS-IoT/blob/master/targets/TARGET_NUVOTON/TARGET_NU_IOT_M2354/PinNames.h)

jeromecoutant, 2023-03-08:
Please have a look on
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM
https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM#custom-boards
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/README.md#stm32h745xx

closed 2023-04-12 by 0xc0170